ArduPilot/ardupilot #28767: Copter: Internal Errors 0x100000 L:57 flow_of_ctrl with onboard computer automatic take-off

afonsoVale, 2024-11-28:
## Bug report

**Issue details**

When performing an automatic take-off using onboard computer commands in GUIDED mode, ArduCopter throws an internal error 0x100000. 

**Version**
ArduCopter V4.5.7

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Hexacopter (DJI F550 frame)

**Hardware type**
CubeBlack flight controller, Here+ V2 GPS, NVIDIA Jetson Xavier NX onboard computer connected via SERIAL2

**Logs**
[log_19_2024-11-19-16-31-18.bin.zip](https://github.com/user-attachments/files/17947568/log_19_2024-11-19-16-31-18.bin.zip)

I believe this issue is being caused by the GUIDED mode SubMode not being set to take-off, as the automatic take-off was being performed by an external controller. Should this be allowed, or should users use the ArduCopter automatic take-off feature instead? In our use case, we are able to activate our external controller in mid-air anyway, which would remove the need for this, but I suppose some use cases might want to experiment with external controllers for take-off.

rmackay9, 2024-11-29:
Hi @afonsoVale,

The internal error is coming from [this line in the landing detector](https://github.com/ArduPilot/ardupilot/blob/Copter-4.5/ArduCopter/land_detector.cpp#L57) and I think we've probably fixed the issue in Copter-4.6 with [this PR](https://github.com/ArduPilot/ardupilot/pull/28129).

I wonder if you'd be happy to re-test with 4.6.0-beta1?  It's an early beta so there is a chance that it has issues but developers have been flying it so my guess is that it is safe.  Another option is to back-port the fix to 4.5.  It's probably quite straight forward if you're happy to re-compile AP.  If you're not then I could create a binary for you.

You could also get around the issue by changing how the companion computer attempts to control the vehicle.  Currently GUID_OPTIONS = 8 so the external computer's SET_ATTITUDE_TARGET message's Thrust field are interpreted as actual thrust instead of climb rates.  That matches the MAVLink spec but most people find it easier to control using climb rates

afonsoVale, 2024-11-29:
Hi @rmackay9,

Thank you for the response. I will have a try with the 4.6.0-beta1 and I'll get back to you.  I'm happy to compile AP as well, so I can integrate the PR in the current stable version as well, thank you for the help! From looking at the PR, I am unsure if it will fix it as I would imagine in our scenario set_angle() is never called with use_thrust = False, but I will for sure have a try.

About changing how the companion computer control the vehicle, I agree. Unfortunately in this application we have some MSc and PhD students who are working with MRS (https://ctu-mrs.github.io) for their higher level applications, and unfortunately this MRS framework relies on using thrust commands only rather than climb rate, although in my opinion using climb rate would be preferable as it it would probably be safer and more stable too. (Note that MRS doesn't claim to be compatible with ArduPilot, but we have made the effort to do some slight modifications to it in order to support ArduPilot, as in the end it is using mavros anyway)

afonsoVale, 2024-12-12:
Hi @rmackay9,

Went ahead and tested today with 4.6.0-beta2, and it looks like the issue is fixed and no internal error is thrown now with this onboard computer controller automatic take-off. 

Log: 
[log_121_2024-12-12-12-46-10.bin.zip](https://github.com/user-attachments/files/18115976/log_121_2024-12-12-12-46-10.bin.zip)

I think we can close this issue now.

Many thanks for the help!

closed 2024-12-12 by afonsoVale

rmackay9, 2024-12-12:
@afonsoVale,

Great!  thanks for the feedback!