ArduPilot/ardupilot #28769: Vehicle does not move and “link down” problems sitl json and jsbsim

Saleh-At, 2024-11-28:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

I ran software-in-the-loop (SITL) simulation with the -f JSON option, and I sent my data based on the JSON format described in the README file, using My FDM. However, I encountered a “GPS unhealthy” error. To resolve this, I set the ahrs_ekf_type parameter to 10, and the error was fixed.

Despite this, although the pn, pe, and pd values were changing, the vehicle itself did not move in Mission Planner (MP). In the HUD, the values were being updated, but on the map, the vehicle remained stationary, which was puzzling and confusing. And displayed me packet lost too much in the console window.

Now, I want to know if it is possible to send the position vector as LLA (latitude, longitude, altitude) to ArduPilot, and if so, which file needs to be modified to implement this.

My second question is about JSBSim simulation. I followed your documentation precisely, but I repeatedly experienced a “link down” error, with the link disconnecting for about 15 seconds, reconnecting for 2 seconds, and then disconnecting again for another 15–20 seconds. My JSBSim version is 1.2.1, and I tested the latest version (1.2.2) as well, but the issue persisted. but in console window I dont have any delayed and packet lost and this is more confusing :face_with_thermometer:

Both of these simulations were performed on Windows using Cygwin.
After restart my win 11, at the start in console window => link1 down and did not run jsbsim.exe
I'm SO tired.
Can I run jsbsim.exe individually (like jsbsim team explains) and ardupilot  and with `fdm_port = 5504` connect together

**Version**
4.7

**Platform**
[  ] All
[  ] AntennaTracker
[ * ] Copter
[ * ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Rascal

**Hardware type**
SITL

**Logs**

IamPete1, 2024-11-30:
> Now, I want to know if it is possible to send the position vector as LLA (latitude, longitude, altitude) to ArduPilot, and if so, which file needs to be modified to implement this.

No, this is not possible. Instead you set the starting location with lat, long and alt the interface then allows positions relative to that starting location.

Saleh-At, 2024-11-30:
@IamPete1 How to convert LLA (lat lon alt) to position(north, east, down) (m) earth frame for [JSON](https://github.com/ArduPilot/ardupilot/tree/master/libraries/SITL/examples/JSON)