ArduPilot/ardupilot #28775: Plane 4.6 beta - Takeoff throttle behavior for planes without airspeed sensor changed from 4.5.7

menschel, 2024-11-29:
## Bug report

In my application forced max throttle regardless of climb angle results in
- Over-speeding of the plane if climb angle is low
- Over-stressing the battery if there are constraints on the discharge curve.

**Issue details**
There have been design changes with Plane 4.6 that change takeoff throttle behavior for planes without an airspeed sensor.
Throttle is forced to max throttle while it was TECS controlled based on climb angle in Plane 4.5.7.

With the ZOHD Drift that I for recreational purposes and development, this breaks at least 2 constraints of the Plane.

- It has a recommended max speed of 50km/h or about 15m/s. This is well exceeded to 60km/h or 17km/h. There is also the suspicion that this over-speeds the wing profile, causing it to loose lift.

- It has a custom Li-Ion battery based on 18350 cells that are rated for 3A (5A burst for 10s).
With Plane 4.5.7, it was possible to adjust the current during takeoff by adjusting the takeoff pitch, i.e. 12° lead to 3A.
This is void with Plane 4.6 because it draws 5.5A for more then 30 seconds, thus violates the discharge profile.

**Version**
4.6beta

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
ZOHD Drift, i.e. Plane, sub 250g 

**Hardware type**
SpeedyBee F405 Wing Mini

**Logs**
[24-11-18_12-43-12.bin](https://github.com/menschel/ardupilot/blob/stash/build/bin/Arduplane/24-11-18_12-43-12.bin)

Georacer, 2025-01-21:
We intentionally changed the throttle profile in airspeedless takeoff here: https://github.com/ArduPilot/ardupilot/pull/27174/files#diff-499a0c85162572f911764247ca98ae60042d4b6e52ba693b7fdcf9173532810b

And will document it here: https://github.com/ArduPilot/ardupilot_wiki/pull/6149/files#

We were discussing this in a thread which ends here: https://github.com/ArduPilot/ardupilot/pull/27174#issuecomment-2245498008. Although it can be a bit confusing as `TKOFF_LVL_ALT` is also mentioned there.

I think this is a good change, because in 4.5 TECS wouldn't actually control airspeed during airspeedless TAKEOFF anyways. It would:
1 Climb at full throttle up until `TKOFF_LVL_ALT` (5m by default).
2 Then it would switch to normal TECS flying, during its climb to `TKOFF_ALT`. During that stage, it would:
  1 operate under speedweight=0
  2 control throttle according to airspeedless laws (pitch-based)
  3 still risk stalling, depending on your aircraft. Primarily because you can't guarantee that you have accumulated enough airspeed past `TKOFF_LVL_ALT`.

rmackay9, 2025-02-17:
Based on @Georacer's comment I'm going to go ahead and close this.

@menschel, are you OK with this?  Thanks again for helping out with the beta testing

closed 2025-02-17 by rmackay9

menschel, 2025-02-18:
Hello @rmackay9,

I'm no longer active on github due to the recent change of github's 
policy. They decided to purge users, that fit certain criteria but don't 
consent to their new rules, obviously on Microsoft's behalf.

Concerning the issue, I consider it a degradation of functionality that 
will eventually cause problems down the road.
I disagree with @Georacer's analysis on a technical basis. We've been 
though this discussion multiple times now.
By observation TECS does control throttle in 4.5.7 and throttle results 
roughly in ((ATT.Pitch / TECS_PITCH_MAX) * THR_MAX). Whatever the math 
is behind this, it is the outcome of the regular TECS tuning process.
If we disagree on that, I can't help it.

The workaround to cope with this degradation in 4.6 is to set takeoff 
altitude extremely low (10m) and set a regular waypoint to the intended 
takeoff destination with the intended altitude that results in the pitch 
value, that would normally be provided in the takeoff mission item. This 
also counters crosswind perfectly.

BR