ArduPilot/ardupilot #28794: Compiling bootloader from source issues

davidkhan, 2024-12-01:
Hello , this is first time i am posting and not sure the right place.

   I have flight controller boards 
MicoAir743 [ardupilot/libraries/AP_HAL_ChibiOS/hwdef/MicoAir743/hwdef.dat at master · ArduPilot/ardupilot · GitHub](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/hwdef/MicoAir743/hwdef.dat)

MicoAir405v2 [ardupilot/libraries/AP_HAL_ChibiOS/hwdef/MicoAir405v2/hwdef.dat at master · ArduPilot/ardupilot · GitHub](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/hwdef/MicoAir405v2/hwdef.dat)

Pixhawk6C MINI [ardupilot/libraries/AP_HAL_ChibiOS/hwdef/Pixhawk6C/hwdef.dat at master · ArduPilot/ardupilot · GitHub](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/hwdef/Pixhawk6C/hwdef.dat)

I was trying to build bootloader on my MacOS machine , all compile ok and even .bin sizes are same. I use STM32CubeProgrammer , full erase and upload , but doesn't work.
Doesn’t work i mean , no led flashing , no hardware discovery on my windows and MAC.
Though it could be build system , I tried on my linux ubuntu 24 machine and upload , but still the same.

   I tried to build bootloaders for all board and tested and none of the board work. 
   I have attached 2 bootloaders , one is my build and other one is original.
***** Please note all original bootloaders under [ardupilot/Tools/bootloaders at master · ArduPilot/ardupilot · GitHub](https://github.com/ArduPilot/ardupilot/tree/master/Tools/bootloaders) are working fine. and i tried from latest sources and older as well*******

Any idea what it could be and have anybody same issues
[BOOTLOADER.zip](https://github.com/user-attachments/files/17970592/BOOTLOADER.zip)
?

Regards,
David

rossiangela, 2024-12-20:
I see several important differences between your build and the official bootloader:

**Size Difference**:
- Your build: 19,681 bytes
- Official build: 19,571 bytes
- Difference of 110 bytes

Content Differences:

18,473 bytes differ between the two files
First difference appears very early at offset 9
This suggests fundamental differences in the compiled code

**Early Differences:**
The differences start in what appears to be the vector table or early initialization code, this could indicate differences in:

- Compiler settings
- Linker configuration
 Memory layout
- Build flags

Based on my analysis, here are some potential issues to check:

**Build Environment:**
- Ensure you're using the exact same compiler version as the official builds
- Check that all build flags match the official configuration
- Verify the optimization levels (-O0, -O2, etc.)

**Toolchain Setup:**
- Confirm you're using the correct ARM toolchain
- Verify the compiler prefix (e.g., arm-none-eabi-)
- Check if any environment variables might be affecting the build

**Hardware Definitions:**
- Double-check the board-specific hwdef.dat files
- Verify all pin configurations match
- Ensure memory layout definitions are correct

**Could you share:**
- The exact build commands you're using?
- Your compiler version (arm-none-eabi-gcc --version)?
- The content of your wscript or build configuration?

**MY ANALYSIS**

```
`// Read both bootloader files
const myBuild = await window.fs.readFile('AP_Bootloader_mybuild_linux.bin');
const official = await window.fs.readFile('MicoAir405v2_AP_Bootloader.bin');

// Compare file sizes
console.log('My build size:', myBuild.length, 'bytes');
console.log('Official size:', official.length, 'bytes');

// Look for major differences by comparing chunks
let differences = 0;
let firstDiffPos = -1;

for (let i = 0; i < myBuild.length && i < official.length; i++) {
    if (myBuild[i] !== official[i]) {
        differences++;
        if (firstDiffPos === -1) firstDiffPos = i;
    }
}

console.log('Number of different bytes:', differences);
console.log('First difference at position:', firstDiffPos);

// Look at the context around first difference
if (firstDiffPos !== -1) {
    const start = Math.max(0, firstDiffPos - 8);
    const end = Math.min(myBuild.length, firstDiffPos + 8);
    
    console.log('\nContext around first difference:');
    console.log('My build:', Array.from(myBuild.slice(start, end)).map(b => b.toString(16).padStart(2, '0')).join(' '));
    console.log('Official:', Array.from(official.slice(start, end)).map(b => b.toString(16).padStart(2, '0')).join(' '));
}`
```

ashwanisihag, 2025-03-04:
Hi Rossi

 i have the same issue , can you please share the official compile  environment?

Regards

peterbarker, 2025-04-29:
@davidkhan that is strange.  I can't speak to the MacOS problems, but you should be able to get this working on your Linux machine with a minimum of fuss.  Did you install your Linux dev env using the standard environment install scripts - and verified that you're using the 10.2.1 compiler?

peterbarker, 2025-10-21:
No answer in 6-ish months.  Please re-open if you have still have issues.

closed 2025-10-21 by peterbarker