ArduPilot/ardupilot #28797: Serious problem: Crossfire receiver with Plane 4.6.0 beta1: no reconnection after failsafe

dxs94, 2024-12-02:
## Bug report

**Issue details**

Crossfire nano RX, Micro TX V2, TX16S, FW 6.36.
Matek 405 Wing board.
Plane was fine on 4.5.5 beta1, no failure. Flashed 4.6.0 beta1 (also tried latest as of 02 dec 2024), the RC link works as expected. However triggering failsafe by turning radio OFF then ON results in receiver reconnecting as expected (green LED on receiver plus basic telemetry recovered on radio side), however there is no link between flight controller and receiver: no command received from radio, no extended telemetry. Regaining control from radio is impossible and would result in a crash in real life. Rebooting the flight controller restores the link.

**Version**
4.6.0 beta1 and latest (02 dec 24)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Regular plane

**Hardware type**
Matek F405 Wing

**Logs**
None

andyp1per, 2024-12-03:
I've reproduced this on tracer.

rmackay9, 2024-12-03:
Great, thanks for the report, I've added this to [the 4.6 issues list](https://github.com/ArduPilot/ardupilot/issues/28612)

andyp1per, 2024-12-04:
This appears to be caused by https://github.com/ArduPilot/ardupilot/pull/26880

closed 2024-12-05 by tpwrules