ArduPilot/ardupilot #28798: Python 3.12 - No module named ‘imp’

PeterJBurke, 2024-12-03:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**
On Ubuntu 24.10, Ardupilot will not install using the 
install-prereqs-ubuntu.sh
script.

_Please describe the problem_
It appears the imp module has been deprecated and removed in python 3.12, the default version of python in Ubuntu 24.04.
This makes Ardupilot not compile.
Can the developers replace "imp" with "mportlib" in the compile code?

**Version**
_What version was the issue encountered with_
Arducopter 4.5.7

**Platform**
[x  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_
On a brand new Ubuntu instance on Amazon (fresh out of the box), I typed in these commands: (following the directions here https://ardupilot.org/dev/docs/building-setup-linux.html)
    1  sudo apt-get update
    2  sudo apt-get install git
    3  sudo apt-get install gitk git-gui
    4  git clone --recurse-submodules https://github.com/your-github-userid/ardupilot
    5  cd ardupilot
    6  git clone --recurse-submodules https://github.com/peterjburke/ardupilot
    7  cd ardupilot
    8  Tools/environment_install/install-prereqs-ubuntu.sh -y
    9  . ~/.profile
   10  ./waf configure --board MatekH743
   11  history

Then I get this error:
./waf configure --board MatekH743
/home/ubuntu/ardupilot/modules/waf/waflib/Utils.py:733: SyntaxWarning: invalid escape sequence '\d'
  return re.split('\d+$', s)[0]
/home/ubuntu/ardupilot/modules/waf/waflib/ConfigSet.py:14: SyntaxWarning: invalid escape sequence '\ '
  re_imp = re.compile('^(#)*?([^#=]*?)\ =\ (.*?)$', re.M)
/home/ubuntu/ardupilot/modules/waf/waflib/ansiterm.py:267: SyntaxWarning: invalid escape sequence '\['
  ansi_tokens = re.compile('(?:\x1b\[([0-9?;]*)([a-zA-Z])|([^\x1b]+))')
/home/ubuntu/ardupilot/modules/waf/waflib/Context.py:610: SyntaxWarning: invalid escape sequence '\_'
  """
Traceback (most recent call last):
  File "/home/ubuntu/ardupilot/modules/waf/waf-light", line 166, in <module>
    from waflib import Scripting
  File "/home/ubuntu/ardupilot/modules/waf/waflib/Scripting.py", line 10, in <module>
    from waflib import Utils, Configure, Logs, Options, ConfigSet, Context, Errors, Build, Node
  File "/home/ubuntu/ardupilot/modules/waf/waflib/Configure.py", line 16, in <module>
    from waflib import ConfigSet, Utils, Options, Logs, Context, Build, Errors
  File "/home/ubuntu/ardupilot/modules/waf/waflib/Options.py", line 14, in <module>
    from waflib import Logs, Utils, Context, Errors
  File "/home/ubuntu/ardupilot/modules/waf/waflib/Context.py", line 9, in <module>
    import os, re, imp, sys
ModuleNotFoundError: No module named 'imp'

PeterJBurke, 2024-12-11:
Update:
I noticed that the Docker compile file still points to Ubuntu 22.04.
I used Ubuntu 22.04 stock (also available as an instance from Amazon EC2), and the compile worked. (Not docker, just straight compile). So that is a workaround for now, to use Ubuntu 22.04 instead of Ubuntu 24.04.

yuri-rage, 2025-03-06:
@Ryanf55 this does not appear to be a bug but rather a user misconception regarding use of Python venv.

Tested just now with Ubuntu 24.04 and had zero issues.

@PeterJBurke's step 9 should either be `source ~/.profile` or log out and start a new bash session.

peterbarker, 2025-04-29:
Agree with @yuri-rage - that step 9 is wrong.

I believe we've fixed the `imp` problems in ArduPilot, so closing this one now.

closed 2025-04-29 by peterbarker