ArduPilot/ardupilot #28799: Copter: AutoTune produces ridiculously small ACCEL_MAX values

mbuzdalov, 2024-12-03:
## Bug report

**Issue details**

In my 2" whoop-sized quad, running AutoTune produces nice enough rate and angle PIDs, but too small maximum accelerations.

Example: `ATC_ANG_RLL_P` of 25+, but `ATC_ACCEL_R_MAX` of around 1000, written as 4000 because it was clamped from below.

These values are not safe to fly.

**Version**

GitHub master, commit 3ed6ada2e316c52ee9ebae4202c99c681e891693.

**Platform**
[  ] All
[  ] AntennaTracker
[X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

Quadcopter.

**Hardware type**

FlywooF405S-AIO

**Logs**

Two logs are attached, for roll and yaw. They are filtered to 5Hz because the flash chip is very small, but should likely be enough for analysis.

[24-12-03_22-19-27_roll.bin.gz](https://github.com/user-attachments/files/18000663/24-12-03_22-19-27_roll.bin.gz)
[24-12-03_22-56-20_yaw.bin.gz](https://github.com/user-attachments/files/18000664/24-12-03_22-56-20_yaw.bin.gz)

tpwrules, 2024-12-03:
This is reproducible in SITL, on 4.5 `ATC_ACCEL_(R|P)_MAX` are around 500k but on master they are 20k.

mbuzdalov, 2024-12-04:
#28800 fixes this issue nicely.

PR #28800 (2024-12-05): AC_AutoTune: fix tracking of maximum angular acceleration
Issue introduced in https://github.com/ArduPilot/ardupilot/pull/27370 and partially fixed in https://github.com/ArduPilot/ardupilot/pull/27762, though evidently not properly tested. Fixes #28799 .

Failing to track the maximum can result in dangerously low values being calculated for `ATC_ACCEL_[RPY]_MAX` and the vehicle becoming unflyable.

Make the variable a reference so that the maximum value is preserved between function calls.

I have tested tuning in SITL, though I have not tested flyability and I have not tested on a real vehicl [... cut]
closed 2024-12-05 by rmackay9