ArduPilot/ardupilot #28807: Missing/ empty/ extra ESC telemetry log data

MarbleJames, 2024-12-05:
## Bug report

**Issue details**
I have a Quadplane with 5 esc’s connected. Since updating from 4.4.4, extra empty ESC’s (up to 8 in total) have been appearing in logs and the cruise ESC data is often empty.

On occasion data for 5 esc's has been observed in mission planner but only data for 4 has been logged.
[ESC telem logs.zip](https://github.com/user-attachments/files/18020578/ESC.telem.logs.zip)

Just verified to be a firmware issue as follows;

Flashed back to 4.4.4 - 5 esc’s correct data
Flashed 4.6.0 - 8 esc’s 4 with data 4 without
Flashed to 4.4.4 - 5 esc’s correct again.
I haven’t tried all the firmware releases in-between but 4.5.4 and 4.5.0 have also been shown to not work.

(https://discuss.ardupilot.org/t/esc-telemetry-phantom-escs/127106)

**Version**
Post 4.4.4

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
4+1 quadplane

**Hardware type**
Cube orange plus bdshot, 4x apd80, 1x apd120

**Logs**
Two logs included, both same params one 4.4.4 and one 4.6.0
https://drive.google.com/drive/folders/17_GhgOKQxd0-tNSXUfivTVYeqpDNg5mY?usp=sharing

andyp1per, 2024-12-05:
This is a change in 4.5 to support dshot on IOMCU. ESC numbers are now aligned with servo numbers.

MarbleJames, 2024-12-05:
> This is a change in 4.5 to support dshot on IOMCU. ESC numbers are now aligned with servo numbers.

Thanks for the reply, I don’t understand how that affects this though, as far as I understand IOMCU is only the first 8 channels of cube orange? I’m running esc’s on 9-13. Anywhere i can read more on this.

I can deal with empty ESC’s in logs but surely this shouldn’t stop recording the telem stream for the cruise ESC? (especially when i'm getting live telemetry for the 5th in mission planner)

v4.4.4
RC9 - ESC[0]
RC10 - ESC[1]
RC11 - ESC[2]
RC12 - ESC[3]
RC13 - ESC[4]

v4.6.0
RC9 - ESC[1]
RC10 - ESC[2]
RC11 - ESC[3]
RC12 - ESC[4]
RC13 - No telem
? ESC[8]
? ESC[9]
? ESC[10]
? ESC[11]

andyp1per, 2024-12-09:
It should be:

v4.6.0
RC9 - ESC[8]
RC10 - ESC[9]
RC11 - ESC[10]
RC12 - ESC[11]

You shouldn't be getting logging from ESC[0]-ESC[3] if you are not using RC1-4

andyp1per, 2024-12-09:
There may be a logging issue here with S1-S8 if they are not being used

MarbleJames, 2024-12-09:
> There may be a logging issue here with S1-S8 if they are not being used

Just to be clear, RC1-8 are being used for normal PWM servo outputs - so yes shouldn't be producing ESC telem data

rmackay9, 2024-12-09:
I've added this to the 4.6 issues list although it sounds like the issue is in 4.5 as well which means it's not a blocker for the release (but it would probably still be good to resolve if we can)

andyp1per, 2024-12-09:
I actually can't reproduce this with your setting. I found a bug - https://github.com/ArduPilot/ardupilot/pull/28833 but I think EDTv2 is only is 4.6 so don't think that can be your issue. Also CubeOrange doesn't support bdshot, so there can by no telemetry coming from IOMCU. So I am rather at a loss to say what is going on here. I correctly get a log with telemetry on 9-12 only.

andyp1per, 2024-12-09:
If you are able to reproduce this on a cut down config that would help me narrow down.

MarbleJames, 2024-12-10:
Ok made a simple setup, to test. 4x apd  f80 connected to 9-12 and one f120 to rc13, telemetry connected to serial 3 -nothing else connected. Using a cube orange plus on airbot mini carrier - i thought bshot was supported if running bdshot fw (which i am)?

Getting the same behaviour (logs in the drive link above).
4.4.4 - ESC[0],[1],[2],[3],[4] populated 
4.6.0 - only logging 4 ESC's again as shown in photo but seeing 5 in mission planner
[image]
[image]

andyp1per, 2024-12-10:
> i thought bshot was supported if running bdshot fw

Sorry, yes - just not supported on the main channels - only aux

andyp1per, 2024-12-10:
> Ok made a simple setup, to test. 4x apd f80 connected to 9-12 and one f120 to rc13, telemetry connected to serial 3 -nothing else connected. Using a cube orange plus on airbot mini carrier - i [...]

Can you post your parameters for the test? How did you test?

MarbleJames, 2024-12-10:
Params for both FW's in the drive link, couple of small differences when comparing (nothing to do with dshot/serial etc) presumably just fw version related. 
Testing I'm just checking I'm getting telem from all the esc's in mission planner then enabling log disarmed, for a while, disable and then download logs.  I Get the same behaviour in flight etc this just keeps the files a bit smaller.

andyp1per, 2024-12-10:
> Params for both FW's in the drive link, couple of small differences when comparing (nothing to do with dshot/serial etc) presumably just fw version related. Testing I'm just checking I'm getting [...]

Are you arming? Running motor test?

MarbleJames, 2024-12-10:
> Are you arming? Running motor test?

No have just been looking at temp/ voltage fields which are available all the time so no need to arm.

andyp1per, 2024-12-10:
> No have just been looking at temp/ voltage fields which are available all the time so no need to arm. Getting the same behaviour with these log disarmed tests as was in flight.

Oh wait, when you say "serial" you mean that's how you are getting the ESC telemetry?

MarbleJames, 2024-12-10:
> > No have just been looking at temp/ voltage fields which are available all the time so no need to arm. Getting the same behaviour with these log disarmed tests as was in flight.
> 
> Oh wait, when you say "serial" you mean that's how you are getting the ESC telemetry?

Yes getting esc telemetry over serial. (protocol 16)

andyp1per, 2024-12-10:
Still can't reproduce. Can you try 4.6beta1 I think it's likely it has been fixed, just don't know by what.

andyp1per, 2024-12-10:
sorry, try 4.7

andyp1per, 2024-12-10:
It's also possible this is an issue with APD I suppose

MarbleJames, 2024-12-10:
[UAV log viewer](https://plot.ardupilot.org/#/)

andyp1per, 2024-12-10:
This is what I get in my test with 4.7
[log4.zip](https://github.com/user-attachments/files/18082784/log4.zip)

andyp1per, 2024-12-10:
Please can you try with the firmware I am using:
[arduplane.zip](https://github.com/user-attachments/files/18082804/arduplane.zip)

MarbleJames, 2024-12-10:
> What are you graphing with?

Just tried viewing the same log as plotted above in the log browser built into mission planner and i'm getting logs for all 5 esc's there, so there might be a separate issue here as well. 

This doesn't solve the empty ESC issue though.
[image]

andyp1per, 2025-01-04:
Please try this firmware:

[arduplane.zip](https://github.com/user-attachments/files/18306477/arduplane.zip)

rmackay9, 2025-01-13:
Hi @MarbleJames,

We were wondering if you'd had a chance to test the new firmware from @andyp1per linked above?  If possible we'd like to resolve this before 4.6 is released as the stable version and it will give us more confidence if you find that it works

MarbleJames, 2025-01-13:
Sorry haven't yet, ill try to give it a go later today.

MarbleJames, 2025-01-13:
Ok just tested this, and it seems to of broken ESC telemetry for me completely completely.

Running 4.6.0beta 1 - 5 ESC's live telem in MP
Flashed to yours's - no telem at all
Back to 4.6.0beta1 -  5 ESC's live telem in MP
Flashed to yours's - no telem at all.

Rebooted cube and esc's multiple times between each flash and no param changes between.

Sorry if this is unhelpful.

andyp1per, 2025-01-13:
> Ok just tested this, and it seems to of broken ESC telemetry for me completely completely.
>
> Running 4.6.0beta 1 - 5 ESC's live telem in MP Flashed to yours's - no telem at all Back to 4.6.0beta1 - 5 [...]

Please can you post a log

MarbleJames, 2025-01-13:
Ok just run again to get logs and the issue - logs numbered 217 and 218 in my google drive link above.

I had ESC number 1-5 displayed in mission planner and the new FW now displays as 9-13 so the values were actually there!! (i didn't expect the numbering change)

In addition there were no more empty esc numbers so happy to report this as fully fixed.

Only issue remaining (separate from this is) https://plot.ardupilot.org/#/ will only display data for 4 ESC's whereas https://plotbeta.ardupilot.org/#/ displays data from all 5 looking at the same log. - not sure where to report this but not a mojor issue no I know.

andyp1per, 2025-01-13:
> I had ESC number 1-5 displayed in mission planner and the new FW now displays as 9-13 so the values were actually there!! (i didn't expect the numbering change)

Right, that's expected

rmackay9, 2025-01-13:
@MarbleJames, Thanks so much for testing!!

closed 2025-01-14 by tridge