ArduPilot/ardupilot #28823: New cork check fails on dev4.7 build on autopilots

Hwurzburg, 2024-12-07:
#27029 causes flow_of_control errors on dev Plane build for KakutH7Wing

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

andyp1per, 2024-12-08:
Hi @Hwurzburg - can you possibly post your parameters. Did this occur in flight or on the ground?

robustini, 2024-12-08:
Same on Matek H743 Wing.

Hwurzburg, 2024-12-08:
> Hi @Hwurzburg - can you possibly post your parameters. Did this occur in flight or on the ground?

prearm fail...your new PR fixes (on the bench)

andyp1per, 2024-12-09:
#28828 is an alternative

timtuxworth, 2024-12-09:
I'm getting this on the ground on a QioTek Zealot H743. Here's a log:
https://www.dropbox.com/scl/fi/qpnslilgvkrxnoom6i8tr/log_68_2024-12-9-09-59-34.bin?rlkey=5aswmo6eawgg5n6xj0kwfoc7r&st=5d9b3go5&dl=0