ArduPilot/ardupilot #28856: Firmware updates over Ethernet

rmackay9, 2024-12-14:
With AP's improved Ethernet support we have more flight controllers connected via Ethernet to companion computers.  It would be very convenient for users (and potentially companies with fleets of AP vehicles) if we could support over-the-air (OTA) updates by allowing the flight controller's firmware to be updated over ethernet.

I discussed this with @tridge and he said that this is already possible for AP_Periph devices so much of the code is already there.  We need to be careful though because this will involve updating the bootloader and we need to ensure we don't run out of space

timtuxworth, 2024-12-14:
It seems like a good idea, but also raises security questions. I'd think OTA might need to at least have an option to require signed firmware, which should probably be the default unless explicitly turned off.

rmackay9, 2025-04-16:
The basis of this could be this firmware uploader from our friends at SkyBrush https://github.com/skybrush-io/ap-firmware-uploader