ArduPilot/ardupilot #28872: Telemetry rate parameters (SR0_, SR1_, SR2_*) jitter or fluctuate after being edited in SITL

ronnuriel, 2024-12-15:
## Bug report

**Issue details**

The telemetry rate parameters (SR0_*, SR1_*, SR2_*) fluctuate unexpectedly after being edited in SITL. For example, SR0_EXT_STAT oscillates between values like 4 Hz and 50 Hz, even after being manually set. This inconsistency makes it difficult to maintain stable telemetry output. The issue is specific to SITL and does not occur with actual hardware.
```
param set SR0_EXT_STAT 50
```

**Version**
AP: ArduPlane V4.7.0-dev (a48ff843)
AP: dbfef1aa0b064bcf9d30ec3ad0886edb

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Airframe type: Fixed Wing

**Hardware type**
Hardware type: SITL (Software-in-the-Loop)

**Logs**
```
FBWA> SR0_EXT_STAT     4.0
param show SR0_EXT_STAT
FBWA> SR0_EXT_STAT     4.0
param show SR0_EXT_STAT
FBWA> SR0_EXT_STAT     50.0
param show SR0_EXT_STAT
FBWA> SR0_EXT_STAT     50.0
param show SR0_EXT_STAT
FBWA> SR0_EXT_STAT     50.0
param show SR0_EXT_STAT
FBWA> SR0_EXT_STAT     50.0
param show SR0_EXT_STAT
FBWA> SR0_EXT_STAT     50.0
param show SR0_EXT_STAT
FBWA> SR0_EXT_STAT     4.0
```

ronnuriel, 2024-12-15:
I ran it with the following cmd:
```
 python sim_vehicle.py -v plane 
```

IamPete1, 2024-12-15:
See https://ardupilot.org/dev/docs/mavlink-requesting-data.html.

The problem it that the ground station is also setup to request data at a given rat. So when you set it to something else the GCS changes it back again.

ronnuriel, 2024-12-15:
> See https://ardupilot.org/dev/docs/mavlink-requesting-data.html.
> 
> 
> 
> The problem it that the ground station is also setup to request data at a given rat. So when you set it to something else the GCS changes it back again. 

I don’t use a GCS. I’m only running python sim_vehicle.py -v plane for simulation purposes.

IamPete1, 2024-12-15:
> I don’t use a GCS. I’m only running python sim_vehicle.py -v plane for simulation purposes.

That command launches MAVProxy unless you run with `--no-mavproxy`.

rmackay9, 2024-12-15:
I'm slightly surprised to see that the parameter value itself is actually updated.  I'd expect a ground station to request changes to he stream rates but I wouldn't expect that to change the parameter values.

ronnuriel, 2024-12-16:
> I'm slightly surprised to see that the parameter value itself is actually updated.  I'd expect a ground station to [request changes to he stream rates](https://ardupilot.org/dev/docs/mavlink-requesting-data.html#using-request-data-stream) but I wouldn't expect that to change the parameter values.

I agree—why would the GCS change the parameters? The default value of the variable is 50, but I’m seeing it fluctuate between 4 and 50 Hz

I will try and test using no mavlink soon...

peterbarker, 2024-12-16:
> I'm slightly surprised to see that the parameter value itself is actually updated. I'd expect a ground station to [request changes to he stream rates](https://ardupilot.org/dev/docs/mavlink-requesting-data.html#using-request-data-stream) but I wouldn't expect that to change the parameter values.

Because Plane persists the streamrates by setting the parameters - have a look at the `persist_streamrates` virtual method: https://github.com/ardupilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Param.cpp#L155

That's not my favourite part of ArduPilot - Plane and Rover persist, none of the other vehicles do.

peterbarker, 2024-12-16:
@ronnuriel I think the things you're seeing are adequately explained?  Ok to close this one?

ronnuriel, 2024-12-16:
> @ronnuriel I think the things you're seeing are adequately explained?  Ok to close this one?
> 
> 

I still didn't get my answer.. I am also changing in my gcs (mission planner) the values and writing it to site but it keeps changes

rmackay9, 2024-12-17:
@peterbarker I'd like to change Rover to be like Copter so that it doesn't persist the stream rate parameters.  Maybe I'll try doing that

rmackay9, 2024-12-17:
I've created a couple of PRs related to this discussion:

1. [Rover stops persisting the requested rates to the parameters](https://github.com/ArduPilot/ardupilot/pull/28894).  This means the parameters will return to their defaults if the vehicle is rebooted
2. [added examples to the Developer wiki's mavlink interface's Requesting Data page](https://github.com/ArduPilot/ardupilot_wiki/pull/6462)

peterbarker, 2024-12-17:
> Hi @ronnuriel,
> 
> mavproxy is changing the rates as well so both GCSs (mission planner and mavproxy) are fighting with each other. I forget how to stop mavproxy from changing the rates though.

`set messagerate -1`

ronnuriel, 2024-12-17:
> I've created a couple of PRs related to this discussion:
>
>
>
> 1. [Rover stops persisting the requested rates to the parameters](https://github.com/ArduPilot/ardupilot/pull/28894).  This means the [...]

What about plane? Can I add a pr for the plane?

rmackay9, 2024-12-18:
@ronnuriel,  We decided that we should leave Plane as-is for now

closed 2024-12-18 by IamPete1