ArduPilot/ardupilot #28876: Matek H743 servos issue on boot sometimes.  Fix exists but needs to be added to ardupilot.

thespova, 2024-12-16:
## Bug report

**Issue details**

Servos go to full deflection on boot some times and it damages hardware.  @andyp1per has already created a fix that I have tested some and it appears to solve the issue. The code just needs to be merged I think so Matek H743 users can go to new versions of ardupilot.

**Version**
4.3.7 through 4.5.7 at least. I assume all previous versions

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
I discovered this on a Traditional Helicopter.  

**Hardware type**
Matek H743 Mini

**Logs**

Previous discussion: https://discuss.ardupilot.org/t/servos-some-times-go-crazy-on-start-up/98641/56
Fix already created by @andyp1per that I have tested some already. Seems to fix the issue.

Please add @andyp1per's code to regular ardupilot code so I can use new versions of ardupilot.  It looks like just a few lines so I hope this is a pretty simple fix.

peterbarker, 2024-12-16:
Link to Andy's branch: https://github.com/andyp1per/ardupilot/tree/pr-gpio-bootloader

@andyp1per - creating a PR for that might make this easier to discuss on DevCall?

andyp1per, 2024-12-16:
Thanks @peterbarker - @tridge didn't like my fix, preferring to leave the pins floating - but its clear that doesn't work in this case. Good discussion to have on the devcall I think

thespova, 2024-12-16:
What is "floating"? There should be no PWM signal on PWM pins until commanded. This makes the Matek H743 series unusable with ardupilot on anything that uses servos.

Thanks @peterbarker for linking @andyp1per 's code. I forgot to add that.

andyp1per, 2024-12-30:
@tridge your thoughts on this appreciated

thespova, 2025-01-20:
Any update on this? What's next?

andyp1per, 2025-01-20:
@thespova I have not had a chance to talk to tridge, but @bugobliterator asserts that floating is the right state for the pins and that anything else is dangerous. If they need pulling up or down that needs to be done in hardware with a resistor. Apparently some servos do this already and this is the reason the CubeOrange has hardware level shifters on the main outputs.

thespova, 2025-01-21:
Id would be nice if that convo with @bugobliterator was captured better here. 

Either way to my understanding I agree that floating is what is best. Just according to my testing that's not what is happening.

My servos go full lock one direction and hold that position. They do not action like some just random signal for a second. They act like they are being commanded to a specific position.

joshanne, 2025-01-21:
I could swear I've seen Tridge commenting about this sort of behaviour, somewhere on discuss maybe? It's been a while since I stumbled across it.

I believe the behaviour is like the pulse width being shortened due to the reboot.

I'd be curious, when it drives to your full deflection, is it in the same direction as SERVOn_MIN?

I've seen some servos, with a usable range of 850-2150us drive to the minimum endpoint when you drive to say, 700us. So if your 1500us pulse is cut short, the servo sees this and drives to the min deflection.

thespova, 2025-01-22:
> ossible, I'd be curious to know what model servo you are using?

SERVOn_MIN direction on the one that has the most issue. I think min on the others too but since I am usually in a panic about the one ripping out its own linkage I pay more attention to that one.

These are Align DS450 servos.

I have a single Align DS455 on the tail and cant recall if the tail also goes nuts. I also use RC passthrough for a cam tilt chan and iv never noticed if it also does the servo freak out (for the same reasons mentioned above), that is a KST X08H V6. 

I have been taking steps to avoid the issue recently like powering the FC from usb before the rest of the Heli but I am building another Heli with the H743Slim and this also uses the same servos for the swash and tail. I might try to power it a few times and see if I can reproduce. I do recall this happening more often when I have recently power cycled the heli AND I have capacitors on both the FC power line and the servo power rail that might mean things are not fully powered down when I start to power them back up.

rmackay9, 2025-01-22:
@thespova,

I think there are two possible issues:

1. on reboot the last PWM signal can be shortened
2. on reboot the PWM pins may momentarily go high and then low again and some servos and ESCs may misinterpret this as a PWM signal and move.  Normally in the AP dev team we blame the servo or ESC for not having enough checks built in

Perhaps @joshanne is referring to the 1st issue while @thespova is experiencing the 2nd?  I'm not sure of course.

If the pull down or pull up ideas mentioned above don't work, another idea might be to introduce an AP_Periph device between the autopilot and servos.  MatekSys sells these and then you'd have more control of what happens at startup.  I've never done this so it's a shot in the dark really

peterbarker, 2025-01-29:
@thespova could you try asserting the thing one way or the other with a very weak pull-up or pull-down resistor?

thespova, 2025-02-09:
> [@thespova](https://github.com/thespova) could you try asserting the thing one way or the other with a very weak pull-up or pull-down resistor?

Can you link me to some instructions on how to do the asserting?

Iv been waiting a little longer on reboots now to try and see if its what @rmackay9 suggested. I do have caps on both the servo power and the board power supply (they are not the same rail) so that actually makes a lot of sense to me if its reboots causing the issue.

rmackay9, 2025-02-10:
Hi @thespova,

To setup a pull down, normally a very high value resistor (like 1000 ohm) is connected to both the signal line and the ground pin.  The high resistor means that just a small amount of power will leak from the signal line.  If there's no actual signal being provided this results in it becoming equal to ground.  If there is a signal provided it will easily overpower the "leak".

hyperion11, 2025-02-14:
may be this is my issue?
https://discuss.ardupilot.org/t/servo-signal-while-tracker-not-connected-to-mav/128740/4

thespova, 2025-02-14:
I think I have determined this is not an issue with the boot loader. On another heli with a Matek H743Slim I have a gimbal and was using the RCIN to directly control the x and y. I noticed that as the board booted up this would put the gimbal in extreme position until the the RC link was fully established and the board fully booted. I have switched that gimbal to RCIN Scaled now and it goes to the trim position (as its supposed to). It goes to this position at the same point in the startup that my other heli moves the swash servos to the extreme and breaks the linkage.  This also makes sense its not the boot loader because the servo keeps pulling on the other heli until the self test starts or until the rc link is established and the board fully booted.

It seems to me there needs to be a setting for what that servo output is that controls the start up position of the servos.  Control surfaces (heli tail and swash plate) should not default to extremes like ESC do.  Just setting it to Moter1 vs ESC in my mind should make this change as it does with weather or not it can moved unarmed.  You can move control surfaces with unarmed and with the motor interlock set.  What ever logic does that should also set the neutral position of the control surfaces it seems like.

Maybe im way off and misunderstanding.

thespova, 2026-10-07:
> Hi [@thespova](https://github.com/thespova),
> 
> To setup a pull down, normally a very high value resistor (like 1000 ohm) is connected to both the signal line and the ground pin. The high resistor value results in just a small amount of power leaking from the signal line. If there's no actual signal being provided this results in it becoming equal to ground. If there is a signal provided it will easily overpower the "leak".

Does the do the same thing as "DEFAULTGPIO OUTPUT LOW PULLDOWN" in the boot loader? If so this is the bootloader fix that @andyp1per  built once before.

I figured out the gimbal thing I was talking about here https://github.com/ArduPilot/ardupilot/issues/28876#issuecomment-2660120461 and it was unrelated. However I still have this issue on my Matek h743 boards.

I found a very similar issue posted in the plane threads https://discuss.ardupilot.org/t/powering-off-servos-while-booting-up-to-prevent-breaking-rods/74066/14

Is this a issue specifically with the Matek H743 possibly?

thespova, 2026-10-07:
Looking around I see a lot of boards using "DEFAULTGPIO OUTPUT LOW PULLDOWN" with notes

"""
# default to all pins low to avoid ESD issues as recommended my STM
--
# wait to enable until every IO is accounted for below.
DEFAULTGPIO OUTPUT LOW PULLDOWN
"""