ArduPilot/ardupilot #28921: Remove AP_Beacon_Mavelmind driver?

rmackay9, 2024-12-20:
[This wiki PR](https://github.com/ArduPilot/ardupilot_wiki/pull/6475) indicates that the existing driver doesn't work and the manufacturer recommends that users instead setup the system as a fake GPS.

I generally think we should go with the manufacturers recommendation so to save flash and reduce confusion I think we should remove the [AP_Beacon_Marvelmind driver](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Beacon/AP_Beacon_Marvelmind.h).

amilcarlucas, 2024-12-20:
I suggest remove ASAP, it adds no value.

marvelmind, 2024-12-20:
Hello colleagues,

I am the CEO of Marvelmind Robotics. My colleague Vladimir pointed me to the discussion. I can hardly contribute anything smart here, but when I see "delete" or "disable", it worries me :-) We don't want something valid - descriptions of steps, drivers, pages - to be accidentally removed because of a misunderstanding of any sort.

We have many ArduPilot users and highly appreciate being indirectly part of the community. Thus, we wanted to make sure that the data and recommended steps are correct and asked for an update yesterday. 

Before removing what has been existing, let us discuss internally and with you how it was implemented and maybe do some tunings to reflect the changes. Again, I can't say whether removing or keeping and updating is better. Let's discuss that and find a commonly accepted solution.

The steps we have published work for sure because we just tested them and wanted to share them with the users to simplify the adoption of ArduPilot + Marvelmind.

Even more - how can we contribute more? Again, I am not familiar with the process. But since many users are coming from your direction, we are very happy to be involved more and closer.

Please comment, and let's agree on the steps.

Thank you.

Kind regards,
Maxim

smoker771, 2024-12-20:
Hello. I'm Vladimir, a software developer from Marvelmind Robotics. If it is possible to keep the [AP_Beacon_Marvelmind driver](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Beacon/AP_Beacon_Marvelmind.h) , I would update it to provide compatibility with latest version of Marvelmind software. Is it possible to join the Ardupilot developers community for this? As I have access to both Pixhawk and Marvelmind hardware and software, hope it would be possible for me to resolve this incompatibility.

rmackay9, 2024-12-21:
Hi @marvelmind, @smoker771,

Great to hear from you!  We won't make any rash moves and the decision would probably need to go through a weekly dev call to get a few more opinions anyway.

In terms of getting more involved, developers are always welcome and if Vladimir needs help on testing in SITL or whatever we have a few [communication channels](https://ardupilot.org/ardupilot/docs/common-contact-us.html) including the Discord server where you'll find various text and voice chats.  A couple of us are often hanging out in the General voice chat.  Just put "@rmackay9" in your text and I'll see.

We also have Tuesday morning (Australia time) weekly dev call and the smaller Wed afternoon EU Dev call where we discuss the details of PRs an the current release.

At a company level, we have [the Partners programme](https://ardupilot.org/copter/docs/common-partners-program.html) in which we setup a private chat with the company and that gives more direct immediate contact with various devs who can provide support.

smoker771, 2024-12-23:
Hello @rmackay9 ,
Thank you very much for your answers and useful links.
I will try to work on updates to return compatibility of the [AP_Beacon_Marvelmind driver](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Beacon/AP_Beacon_Marvelmind.h) with Marvelmind software in close time.
Hope this will be valuable for Ardupilot users as another way  of integration with Marvelmind navigation system.

peterbarker, 2025-04-29:
@smoker771 how goes the work?