ArduPilot/ardupilot #28937: AC_Fence: misleading results from get_breaches() during manual recovery period

ntamas, 2024-12-23:
## Bug report

**Issue details**

The bitmask of active fence breaches returned from `AC_Fence::get_breaches()` is not updated when the vehicle is in a manual recovery period (triggered by a mode switch when at least one fence is actively breached). This has multiple consequences:

* The onboard log is spammed with "fence breach cleared" `ERR` log records on Copter (subsystem = 9, error code = 0). This is due to `AC_Fence::check()` always returning zero (no new fence breaches) -- the logic in `Copter::fence_check()` hits the branch where `orig_breaches != 0` and `new_breaches == 0`, triggering a log entry every 40 msecs.
* Any third-party code that relies on the assumption that `AC_Fence::get_breaches()` always returns the _currently_ active breaches is broken because the breaches are not updated in the recovery period. If the behaviour is intentional, the comment of `AC_Fence::get_breaches()` should be updated to emphasize the existence of the manual recovery period and that it freezes the breach flags.

A preferred fix would probably make sure that `AC_Fence::get_breaches()` is always kept up-to-date and would only defer fence-related failsafe _actions_ until the end of the recovery period, but I cannot judge whether this would have any unintended consequences elsewhere.

**Version**
ArduCopter 4.5.7

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad, but probably not relevant to the issue.

**Hardware type**
Custom, but probably not relevant to the issue.

**Logs**
I am not allowed to disclose the log where we found the issue, but I can provide it upon request to the developers. I can be reached on the ArduPilot and Skybrush Discord servers.

andyp1per, 2024-12-23:
@ntamas are you able to verify on 4.6? I have a feeling I might have fixed this in the fence rework

ntamas, 2024-12-23:
I'll try to test later with 4.6. I just checked the new code in `AC_Fence::check()` on Github and it seems like the issue might be fixed in 4.6 because the early exit of the manual recovery has now been moved to the very end of the function - it still returns 0 (pretending no new breaches) but the side effects of the `check_*()` functions are now executed and `_breached_fences` seems to be updated properly.

peterbarker, 2025-05-09:
Ping @ntamas - retesting?

rmackay9, 2025-05-13:
@ntamas, txs very much for the testing and feedback!

closed 2025-05-13 by rmackay9