ArduPilot/ardupilot #28953: Copter/Rover: maintain speed even when waypoints are placed close together

rmackay9, 2024-12-27:
Copter/Rover SCurves has a limitation where it only looks forward two waypoints (e.g. the current waypoint and the next one) and for safety reasons it plans the maximum speed so that the vehicle can stop in time for the 2nd waypoint.  This can lead to the vehicle unnecessarily slowing if the user places the waypoints close together.  How "close" depends upon the WPNAV_ACCEL and WPNAV_JERK parameters.

We could reduce the user experienced slowdowns by enhancing SCurves to look further ahead.

anand2532, 2025-01-04:
Can you plwase elaborate :
- Exact conditions causing unnecessary speed reduction
- Quantify performance impact with specific waypoint spacing scenarios
- Current SCurves implementation limitations in path planning

xianglunkai, 2025-01-14:
@rmackay9 
Hello! Long time no see! I have three suggestions and questions:
1. Can Rover support both L1 and scurve algorithms, which is convenient for users to choose and compare?
2. At present, Scurve does not support obstacle avoidance and regression reference lines for simple path planning.
3. For the trajectory tracking of the boat, is there any consideration based on the angular velocity instead of the speed course, and the current mainstream is based on the course speed control

rmackay9, 2025-01-14:
Hi @xianglunkai, nice to hear from you.

1. No, L1 is gone I'm afraid and there are no plans to bring it back in future versions.  It's too much development and support effort for our small team to support both L1 and SCurves.  Instead we've chosen to put our limited resources towards supporting and enhancing SCurves only
2. Rover's autonomous modes (including Auto, RTL, etc) do support both simple avoidance and path planning.  I think maybe you're referring to [this issue](https://github.com/ArduPilot/ardupilot/issues/20406) where Rover does not return to the original line?  It would be nice to fix that for sure
3. I don't fully understand the question but SCurves plans the path based on the user configured maximum speed, acceleration and jerk.  It does not consider the vehicle's maximum turn rate.  Turn rate can be calculated from speed and lateral acceleration but the frame's mechanical method of steering the vehicle (e.g. servo, vectored thrust, etc) can mean that you may or may not be able to adequate specify a maximum acceleration that closely matches how the vehicle can actually behave

Hope this helps, sorry if I've misunderstood something