ArduPilot/ardupilot #28956: Plane Offboard Support - Replace GUIDED mode

Ryanf55, 2024-12-27:
## Feature request

**Is your feature request related to a problem? Please describe.**

* Guided mode on plane is behind copter functionality
* GUIDED plane mixes the "FlyTo" GCS functionality with other lower level control commands which are hard such as [yaw rate control over DDS](https://github.com/ArduPilot/ardupilot/pull/26143)
* Guided mode on plane doesn't give a clear API for sending kinematically feasible dubins paths in 3D
* If you need to send multiple lower level guided commands (say, heading and turn rate), the [timeouts interfere](https://github.com/ArduPilot/ardupilot/pull/28526#issuecomment-2456573675) - they assume only a single control methodology at once
* The interface is inconsistent across DDS, Scripting, and MAVLink
* There is a lot of complexity in QuadPlane on what the intent is for certain commands (ie - should you fly in copter mode or in plane mode to accomplish the desired velocity)

**Describe the solution you'd like**

* Make GUIDED mode only for GCS FlyTo
* Create a new mode (name proposed is OFFBOARD) to handle more of these control methodologies using the reserved mode from [here](https://github.com/ArduPilot/ardupilot/pull/28939)
* Allow a user to be able to achieve at least the same capabilities that missions have with position lookahead/s-curves by allowing fly-through positional control
* Allow velocity and acceleration control
* Give offboard/scripting better control for quadplane if it wants
* Allow Lua/DDS/Mavlink to register custom modes that build on "offboard" but call it something different - like "ACRO" or "SMART RTL"

**Describe alternatives you've considered**

* Continue to augment features in GUIDED which is already very convoluted

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Additional context**

* Let's gather some high level requirements on what this offboard mode should accomplish before diving into design/implementation
* Relates to https://github.com/ArduPilot/ardupilot/pull/28930 as discussed in https://discuss.ardupilot.org/t/dev-call-dec-23-2024/127890/2#p-507105-utc-0106-httpsgithubcomardupilotardupilotpull28930pullrequestreview-2521275286-15

**Requirements (WIP)**

* Be able to achieve FlyTo behavior of clicking a waypoint on a map, but levaraging more advanced Lua/DDS path plan validation as needed [here](https://github.com/ArduPilot/ardupilot/issues/28562)
* Be able to control climb rate and turn rate at the same time
* Support position/velocity/acceleration control in body AND earth frame
* Allow a user-defined fallback if the low-level offboard controller fails (like current timeout behavior)
* Continue support of [MAVLink Standard Modes](https://mavlink.io/en/services/standard_modes.html)

timtuxworth, 2024-12-27:
Yes this also reflects my confusion documented in this Wiki PR: https://github.com/ArduPilot/ardupilot_wiki/pull/6096

Ryanf55, 2024-12-30:
More user stories (from users): 
* https://discuss.ardupilot.org/t/switching-between-waypoint-and-heading-speed-altitude-in-guided/118298?u=rfriedman

Ryanf55, 2025-01-06:
This will be nice to be able to get out of GUIDED by a "Click to go" which would work with a  mode change.

timtuxworth, 2025-01-07:
OFBD_ = Offboard

Ryanf55, 2025-05-18:
Proposal:
1. Add new offboard mode, that simply inherits from ModeGuided, with no changes to guided - the only thing new is you can change to that mode in the GCS.
2. Start moving features from ModeGuided into ModeOffboard.,

Would this [PR](https://github.com/ArduPilot/ardupilot/pull/28930) be accepted if we changed it from `plane.control_mode->is_guided_mode()` to `plane.control_mode->supports_external_control()`? Then it means we can have ModeGuided and ModeOffboard return `true` for that and slowly take things out of ModeGuided. Once ModeGuided only supports the FlyTo capability, we change make it return false for  `supports_external_control()`