ArduPilot/ardupilot #28957: Plane: quadplane tilt rate issues

tridge, 2024-12-27:
There are 2 issues with the quadplane tilt down rate for tilt-enabled quadplanes.
- the quadplane.assisted_flight flag oscillates during the TRANSITION_TIMER phase of the fwd transition
- if Q_TRANSITION_MS is smaller than the time it takes to go from Q_TILT_MAX to full tilt with the configured Q_TILT_RATE parameters then the last part will happen at 90 deg/s

timtuxworth, 2024-12-30:
Is this what makes transitioning on a tilt rotor so hmmm - frustrating? It feels like it's trying hard to transition but won't quite get there? I haven't flow my VTBird for a while this is the last flight but I always had this feeling. 
BTW it was always with manually transitioning to FBWA - the auto mode transition always seems to be much smoother and I always wished I could duplicate it. Here's the log:
https://www.dropbox.com/scl/fi/tv8sl0va1xhdti691zrv2/log_154_2024-8-31-18-24-52.bin?rlkey=b7ujssije9iulhdgho9ky603m&dl=0

priseborough, 2025-01-05:
I think there is also a potential issue with the way that TECS can dip the throttle demand at the time that the tilting rotor throttle control hand over to TECS from the motors library. For vehicles where the propwash over the front wing contributes significantly to lift, we should be limiting throttle to no less than the trim value initially.

priseborough, 2025-03-10:
I've created a branch which contains a modification to the forward transition completion criteria to make it wait until rotor tilt forward has completed.

https://github.com/priseborough/ardupilot/tree/pr-quadplane-tilt-rate

I'll have a look at what we can do to prevent a step in throttle for those setups with a front wing  that benefits from prop-wash energisation at lower speeds.

priseborough, 2025-04-06:
See https://github.com/ArduPilot/ardupilot/pull/29696 for fix to throttle glitch.