ArduPilot/ardupilot #28969: Floating Point Exception in the position controller

peterbarker, 2024-12-30:
CI run produced an FPE while running `test.Plane.FlyEachFrame`.  I've never seen this before.

The PR in question is going absolutely nowhere near the relevant code (should actually be a no-compiler-output change in SITL?).  https://github.com/ArduPilot/ardupilot/pull/28966

Link to output is below, also excerpted below.

Seems to be something wrong with the maths in `limit_accel_xy` - we're taking the square root of a negative number; something wrong with taking the cross product, perhaps?!  Thankfully the output below contains values for all of the variables required...

https://productionresultssa10.blob.core.windows.net/actions-results/4e63b159-021f-4b97-a167-86fc7dcf4f13/workflow-job-run-6b0240de-2d48-582d-e68b-9616ef328464/logs/job/job-logs.txt?rsct=text%2Fplain&se=2024-12-30T07%3A30%3A15Z&sig=CO5BGQOMT4%2FEO2n%2F2x77s%2FBvK54kxyiUgE7qSlhKxUs%3D&ske=2024-12-30T19%3A10%3A12Z&skoid=ca7593d4-ee42-46cd-af88-8b886a2f84eb&sks=b&skt=2024-12-30T07%3A10%3A12Z&sktid=398a6654-997b-47e9-b12b-9515b896b4de&skv=2024-11-04&sp=r&spr=https&sr=b&st=2024-12-30T07%3A20%3A10Z&sv=2024-11-04

```
2024-12-30T06:13:01.8519750Z AT-0413.1: AP: VTOL position1 v=21.3 d=169.1 h=49.8 dc=19.6
2024-12-30T06:13:01.8520094Z arduplane: ERROR: Floating point exception - aborting
2024-12-30T06:13:01.8520719Z arduplane: Running: sh Tools/scripts/dumpstack.sh 8040 >dumpstack.sh_arduplane.8040.out 2>&1
2024-12-30T06:13:01.8521189Z arduplane: dumpstack.sh has been run.  Output was:
2024-12-30T06:13:01.8521547Z arduplane: -------------- begin dumpstack.sh output ----------------
2024-12-30T06:13:01.8521872Z arduplane: [New LWP 8041]
2024-12-30T06:13:01.8522144Z arduplane: [Thread debugging using libthread_db enabled]
2024-12-30T06:13:01.8522575Z arduplane: Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
2024-12-30T06:13:01.8523073Z arduplane: 0x00007fe61ddfb42f in wait4 () from /lib/x86_64-linux-gnu/libc.so.6
2024-12-30T06:13:01.8523543Z arduplane: #0  0x00007fe61ddfb42f in wait4 () from /lib/x86_64-linux-gnu/libc.so.6
2024-12-30T06:13:01.8523926Z arduplane: No symbol table info available.
2024-12-30T06:13:01.8524278Z arduplane: #1  0x00007fe61dd61bdb in ?? () from /lib/x86_64-linux-gnu/libc.so.6
2024-12-30T06:13:01.8524638Z arduplane: No symbol table info available.
2024-12-30T06:13:01.8533257Z arduplane: #2  0x000055788a89e19f in AP_HAL::run_command_on_ownpid (commandname=0x55788aa1a1af "dumpstack.sh") at ../../libraries/AP_HAL_SITL/system.cpp:127
2024-12-30T06:13:01.8534641Z arduplane:         command_filepath = 0x7ffdaf31f660 "Tools/scripts/dumpstack.sh"
2024-12-30T06:13:01.8535865Z arduplane:         statbuf = {st_dev = 2065, st_ino = 527244, st_nlink = 1, st_mode = 33261, st_uid = 0, st_gid = 0, __pad0 = 0, st_rdev = 0, st_size = 265, st_blksize = 4096, st_blocks = 8, st_atim = {tv_sec = 1735538476, tv_nsec = 968517313}, st_mtim = {tv_sec = 1735538476, tv_nsec = 968517313}, st_ctim = {tv_sec = 1735538476, tv_nsec = 968517313}, __glibc_reserved = {0, 0, 0}}
2024-12-30T06:13:01.8537613Z arduplane:         custom_scripts_dir_path = 0x0
2024-12-30T06:13:01.8538484Z arduplane:         custom_scripts_dir_path_pattern = 0x0
2024-12-30T06:13:01.8539517Z arduplane:         paths = {0x0, 0x55788aa1a06b "Tools/scripts/%s", 0x55788aa1a07c "APM/Tools/scripts/%s", 0x55788aa1a091 "../Tools/scripts/%s"}
2024-12-30T06:13:01.8541241Z arduplane:         buffer = "Tools/scripts/dumpstack.sh\000\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177"
2024-12-30T06:13:01.8543296Z arduplane:         progname = "/__w/ardupilot/ardupilot/build/sitl/bin\000arduplane\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177"
2024-12-30T06:13:01.8544682Z arduplane:         n = 49
2024-12-30T06:13:01.8545053Z arduplane:         p = 0x7ffdaf31f717 ""
2024-12-30T06:13:01.8546307Z arduplane:         output_filepath = "dumpstack.sh_arduplane.8040.out\000\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177"
2024-12-30T06:13:01.8549709Z arduplane:         cmd = "sh Tools/scripts/dumpstack.sh 8040 >dumpstack.sh_arduplane.8040.out 2>&1\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177"
2024-12-30T06:13:01.8552083Z arduplane:         fd = 2141192192
2024-12-30T06:13:01.8555083Z arduplane:         buf = "\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177\000\000\240\177"...
2024-12-30T06:13:01.8558307Z arduplane: #3  0x000055788a89e34d in AP_HAL::dump_stack_trace () at ../../libraries/AP_HAL_SITL/system.cpp:159
2024-12-30T06:13:01.8559065Z arduplane: No locals.
2024-12-30T06:13:01.8559724Z arduplane: #4  0x000055788a89ef34 in _sig_fpe (signum=8) at ../../libraries/AP_HAL_SITL/SITL_cmdline.cpp:65
2024-12-30T06:13:01.8560456Z arduplane: No locals.
2024-12-30T06:13:01.8560811Z arduplane: #5  <signal handler called>
2024-12-30T06:13:01.8561274Z arduplane: No symbol table info available.
2024-12-30T06:13:01.8561893Z arduplane: #6  0x00007fe61df6c49c in ?? () from /lib/x86_64-linux-gnu/libm.so.6
2024-12-30T06:13:01.8562662Z arduplane: No symbol table info available.
2024-12-30T06:13:01.8563291Z arduplane: #7  0x00007fe61df6e6e5 in ?? () from /lib/x86_64-linux-gnu/libm.so.6
2024-12-30T06:13:01.8563913Z arduplane: No symbol table info available.
2024-12-30T06:13:01.8564666Z arduplane: #8  0x000055788a6d8822 in safe_sqrt<float> (v=-0.03125) at ../../libraries/AP_Math/AP_Math.cpp:73
2024-12-30T06:13:01.8565405Z arduplane:         ret = 566.187256
2024-12-30T06:13:01.8566232Z arduplane: #9  0x000055788a6daa7e in limit_accel_xy (vel=..., accel=..., accel_max=566.187256) at ../../libraries/AP_Math/control.cpp:390
2024-12-30T06:13:01.8567133Z arduplane:         accel_max_dir = 4.59135442e-41
2024-12-30T06:13:01.8567835Z arduplane:         vel_input_unit = {x = 0.984285712, y = 0.176583186}
2024-12-30T06:13:01.8568392Z arduplane:         accel_dir = -0.0795593262
2024-12-30T06:13:01.8568938Z arduplane:         accel_cross = {x = 99.9791489, y = -557.290039}
2024-12-30T06:13:01.8570050Z arduplane: #10 0x000055788a8577ca in AC_PosControl::update_xy_controller (this=0x55788b15f850) at ../../libraries/AC_AttitudeControl/AC_PosControl.cpp:706
2024-12-30T06:13:01.8571094Z arduplane:         ahrsGndSpdLimit = 400
2024-12-30T06:13:01.8571545Z arduplane:         ahrsControlScaleXY = 1
2024-12-30T06:13:01.8572195Z arduplane:         curr_pos = @0x55788ab23238: {x = -17819.6914, y = 3957.76709, z = 4968.15088}
2024-12-30T06:13:01.8572950Z arduplane:         comb_pos = {x = -17819.6914, y = 3957.76709, z = 4968.15088}
2024-12-30T06:13:01.8573522Z arduplane:         vel_target = {x = 0, y = 0}
2024-12-30T06:13:01.8574094Z arduplane:         curr_vel = @0x55788ab23244: {x = 2097.99463, y = 376.385193}
2024-12-30T06:13:01.8574725Z arduplane:         comb_vel = {x = 2097.99463, y = 376.385193}
2024-12-30T06:13:01.8575307Z arduplane:         accel_target = {x = 99.9008408, y = -557.304077}
2024-12-30T06:13:01.8575829Z arduplane:         angle_max = 3000
[... cut ...]
2024-12-30T06:13:22.0842770Z arduplane: -------------- end dumpcore.sh output ----------------
2024-12-30T06:13:22.0843102Z EOF on TCP socket
2024-12-30T06:13:22.0843313Z Attempting reconnect
2024-12-30T06:13:22.0843540Z [Errno 111] Connection refused sleeping
2024-12-30T06:13:22.0843818Z [Errno 111] Connection refused sleeping
2024-12-30T06:13:22.0844228Z [Errno 111] Connection refused sleeping
2024-12-30T06:13:22.0844498Z [Errno 111] Connection refused sleeping
2024-12-30T06:13:22.0844770Z [Errno 111] Connection refused sleeping
```

lthall, 2025-01-01:
It is safe to limit the answer to positve values.
`float accel_max_dir = safe_sqrt(MAX(0, sq(accel_max) - accel_cross.length_squared()));`

tpwrules, 2025-01-01:
It's happened again here: https://github.com/ArduPilot/ardupilot/actions/runs/12572260649/job/35044375709?pr=28857 

Numbers are identical.

closed 2025-01-07 by tpwrules