ArduPilot/ardupilot #29043: Plane: separate mixing gain value for elevons and vtails

robustini, 2025-01-12:
In case of servox_function assigned to elevons or vtails a mixing is applied obviously to handle roll and pitch in the former case and yaw and pitch in the latter case.
The amount of mixing is defined by the MIXING_GAIN parameter.
The problem is that in Ardupilot in case both elevons and vtails are present the mixing goes to affect both.
It would be quite appropriate instead for the mixing to be separameter, and always parameterizable, e.g. “ELEV_MGAIN” and “VTAIL_MGAIN,” or perhaps keep “MIXING_GAIN” but in case of both elevons and vtails act only on the former.
I am trying to implement this with a LUA script but have not succeeded.
Since the functionality is interesting anyway it would be more appropriate in my opinion if it were integrated into the code.

Rolf-G, 2025-01-12:
We encountered a similar problem and solved it (additionally with Crow-Flaps-Implimentation) in LUA-Script like this. Maybe this will help you: https://discuss.ardupilot.org/t/crowflap-lua-mixer-with-differential-spoiler-flying-wing-glider/100250

Rolf

IamPete1, 2025-01-12:
We added a script for this in the end: 
```

-- Setup param block, reserving 2 param beginning with VTAIL_
local PARAM_TABLE_KEY = 75
local PARAM_TABLE_PREFIX = 'VTAIL_'
assert(param:add_table(PARAM_TABLE_KEY, "VTAIL_", 2), 'could not add param table')

-- Add a parameter and bind it to a variable
local function bind_add_param(name, idx, default_value)
    assert(param:add_param(PARAM_TABLE_KEY, idx, name, default_value), string.format('could not add param %s', name))
    return Parameter(PARAM_TABLE_PREFIX .. name)
end

-- Duplicate mixing gain and offset params
local VTAIL_MGAIN = bind_add_param('MGAIN', 1, 0.5)
local VTAIL_MOFF = bind_add_param('MOFF', 2, 0.0)

-- Input servo function numbers
local k_rudder = 21
local k_elevator = 19

-- Use scripting servo function for output
local k_vtail_right = 94 -- k_scripting1
local k_vtail_left = 95 -- k_scripting2

-- Copy the mixing code from AP
-- https://github.com/ArduPilot/ardupilot/blob/609551212afc02f0c0b30fda165f90211eed23df/ArduPlane/servos.cpp#L170-L196
local function update()

    -- the order is setup so that non-reversed servos go "up", and
    -- func1 is the "left" channel. Users can adjust with channel
    -- reversal as needed
    local in1 = SRV_Channels:get_output_scaled(k_rudder)
    local in2 = SRV_Channels:get_output_scaled(k_elevator)

    -- apply MIXING_OFFSET to input channels
    local offset = VTAIL_MOFF:get()
    if offset < 0 then
        in2 = in2 * ((100 - offset) * 0.01)
    elseif offset > 0 then
        in1 = in1 * ((100 + offset) * 0.01)
    end

    local gain = VTAIL_MGAIN:get()
    local out1 = (in2 - in1) * gain
    local out2 = (in2 + in1) * gain
    SRV_Channels:set_output_scaled(k_vtail_right, out1)
    SRV_Channels:set_output_scaled(k_vtail_left, out2)

    return update, 10
end

return update()
```

robustini, 2025-01-13:
Thank you both!

closed 2025-01-13 by robustini