ArduPilot/ardupilot #29045: MatekF405-TE,   SRC_ENABLE,   LUA_SCRIPT,  SCRIPTING_ENABLED.

samwidarr, 2025-01-12:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REQUEST WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please do a careful search before opening this, there are already a lot of feature requests**_
If there was a previous discussion in the forum, link to it

## Feature request

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ 1 ] Plane
[  ] Rover
[  ] Submarine

Please tell me how to add or enable SRC_ENABLE (Lua_script) on matekf405-te, maybe you have a ready firmware or I can add and build just tell me what and where to add, (at your own risk) please help very much need,
Thank you.

Пожалуйста подскажите как можно добавить или включить работу SRC_ENABLE (Lua_script) на  matekf405-te,  может у вас есть готовая прошивка или  я и сам могу добавить и собрать только скажите что и куда нужно добавить, (на свой страх и риск) пожалуйста помогите очень нужно,
Спасибо.

**Additional context**
_Add any other context or screenshots about the feature request here._

IamPete1, 2025-01-12:
You can use the custom build server, you will have to turn off most everything to get scripting to fit. Even then your not going to have a great time on a F405.

https://custom.ardupilot.org/

closed 2025-01-12 by IamPete1

samwidarr, 2025-01-13:
> You can use the custom build server, you will have to turn off most everything to get scripting to fit. Even then your not going to have a great time on a F405.
> 
> https://custom.ardupilot.org/
Thank you very much, but I would like to know where and what to add in the code and what to build myself, as I already use custom firmware (due to the power of vtx),  
I'm actually interested in switching bends and channels, maybe someone can help me with an example:
( rc 6 = 1000 = band 1 chanel 3, 
rc 6 = 1500 = band 3 chanel 1,
rc 6 = 2000 = band 5 chanel 7,).