ArduPilot/ardupilot #29095: Lidar/Rangefinder used ONLY for VTOL takeoff and VTOL landing in Quadplane

Gromitvt10, 2025-01-17:
## Feature request

**Is your feature request related to a problem? Please describe.**
_When flying my quadplane with Lidar installed i enable rangefinder for automatic landing. The rangefinder will be used both on the landing approach and for final flare.  But, if the lidar, for any reason shows a low reading, it will trigger the alt_assist even when not on approach or landing. This in turn initiates the VTOL motors to start and creates a lot of drag. 
The problem is this aircraft can fly for 8 hours, what if it was flying miles away and the lidar randomly starting doing this, false low readings.  It will constantly be turning the VTOL motors on and off and would eventually drain its battery.  We can fix or replace this particular lidar unit, but that doesn't mean the new one won't eventually do the same thing. This issue surfaced only after many flights on the aircraft, who's to say it won't happen again with a new lidar unit after many flights.

**Describe the solution you'd like**
_We currently set the Alt_assist to zero, but that turns it off completely. I would like a way to use the lidar only for VTOL takeoff (transition height-use until transition is complete) and landing measurements (over landing spot all the way until touch down). I would like to use Baro for Alt-assist. I don't need the lidar when i'm flying at 1000ft, especially when it can only read up 300ft.  A setting or flag that lets you chose to "use lidar only for takeoff and landing" would be great.  If not checked then it uses lidar for everything, when checked it uses Baro for Alt_assist.

**Describe alternatives you've considered**
_I have considered not using lidar at all, but then why even have it. Discussed changing the unit or changing unit parameters, this may fix this particular instance but won't fix the overall issue of Ardupilot engaging Alt assist for false readings._

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_This issue has been brought up and discussed in my partner chat_

IamPete1, 2025-01-17:
We do have a Q_OPTION for airspeed assist:
"Disable speed based Qassist when using synthethic airspeed estimates". 

So it would make sense to add this in a similar way. However, I think we have run out of bits.....

timtuxworth, 2025-01-17:
> We do have a Q_OPTION for airspeed assist: "Disable speed based Qassist when using synthethic airspeed estimates".
> 
> So it would make sense to add this in a similar way. However, I think we have run out of bits.....

"You always need more!"

tridge, 2025-01-27:
@Gromitvt10 I think we can do two things:
 - change the parameter RNGFND_LANDING from a 0/1 value to 0/1/2, where value 2 means to use only for landing and not for Q_ASSIST_ALT
 - possibly add a parameter for the maximum discrepancy between rangefinder and terrain altitude from terrain database that we will use the rangefinder. This would apply to landing as well (it could prevent a very early slow down in descent for VTOL landing). I'm thinking users would set it to 50m or so.

peterbarker, 2025-11-18:
Closed by https://github.com/ArduPilot/ardupilot/pull/30458