ArduPilot/ardupilot #29147: Regression in MCU Volt?

MichelleRos, 2025-01-27:
**Issue details**

We seem to have had a regression (or unintentional change?) in the logging of the MCU log message.

What it used to look like (v4.4.0):
[image]
Max and min is since the last logged message.

What it is now (recent master):
[image]
Max and min are since the start of the flight.

This makes it rather less useful for investigating power issues as we don't see the variation in voltage as clearly.

tridge, 2025-01-29:
this is a regression, yes, well spotted!

tridge, 2025-01-29:
could be this one?
```
commit 726dcda24f68da55626cad45b839e9d27d684c04
Author: bugobliterator <siddharthbharatpurohit@gmail.com>
Date:   Tue Jan 17 15:15:15 2023 +1100

    AP_HAL_ChibiOS: add support for using ADC2 and ADC3 on H7
```

MichelleRos, 2025-01-30:
> could be this one?
> 
> ```
> commit 726dcda24f68da55626cad45b839e9d27d684c04
> Author: bugobliterator <siddharthbharatpurohit@gmail.com>
> Date:   Tue Jan 17 15:15:15 2023 +1100
> 
>     AP_HAL_ChibiOS: add support for using ADC2 and ADC3 on H7
> ```

Yup. I was suspecting that commit, but wanted to check first. I just did tests before and after the PR that added this commit (https://github.com/ArduPilot/ardupilot/pull/22671), and confirmed that this PR is what changed it.

peterbarker, 2025-04-29:
Fixed by #29417

closed 2025-04-29 by peterbarker