ArduPilot/ardupilot #29148: Support for Custom Firmware Integration of BMS in ArduPilot Battery Monitor

Shubham07-bit, 2025-01-27:
## Description
I have developed custom firmware to add support for the BMS module in ArduPilot's battery monitor. The changes were based on the SMBus library, and relevant modifications were made in the following files:

### Modifications Made

1. **AP_BattMonitor_config.h**  
   Added macro for the BMS module:
   ```cpp
   #ifndef AP_BATTERY_SMBUS_BMS_ENABLED
   #define AP_BATTERY_SMBUS_BMS_ENABLED AP_BATTERY_SMBUS_ENABLED
   #endif
   ```
2. **AP_BattMonitor.h**
Added a new enum for BMS:
    ```cpp
    BMS = 33,
    ```
3. **AP_BattMonitor.cpp**
Added a case for the BMS module:
    ```cpp
    #if AP_BATTERY_SMBUS_BMS_ENABLED
        case Type::BMS:
            GCS_SEND_TEXT(MAV_SEVERITY_DEBUG, "Set to BMS");
            drivers[instance] = NEW_NOTHROW AP_BattMonitor_SMBus_BMS(*this, state[instance], _params[instance]);
            break;
    #endif // AP_BATTERY_SMBUS_BMS_ENABLED
    ```

The firmware compiled and uploaded successfully, but when testing in ArduPilot, the module does not seem to work. No debug messages (Set to BMS).

## Request for Help

I am seeking assistance with:

    1. Validating the changes made to the codebase.
    2. Troubleshooting why the module is not initializing or working.
    3. Suggestions for additional steps or configurations to ensure successful integration

peterbarker, 2025-01-27:
Try a message level other than DEBUG - it's possible your GCS is filtering anything lower than... something.   Try INFO.

rmackay9, 2025-01-27:
The "Set to BMS" will be called very early in AP's startup so you might miss it

This is really a developer question on an issue so I'm going to go ahead and close it.  Txs for using AP!

closed 2025-01-27 by rmackay9