ArduPilot/ardupilot #29155: Sort parameters with numerics

Ryanf55, 2025-01-28:
## Feature request

**Is your feature request related to a problem? Please describe.**

Docs for params show up in a weird order.

[image]

**Describe the solution you'd like**

see https://github.com/ArduPilot/ardupilot/blob/master/Tools/autotest/logger_metadata/emit_xml.py#L54

**Describe alternatives you've considered**
Manual ordering
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2025-01-28:
Thanks.  Also SERVO10_xxx appears before SERVO2_xxx

LupusTheCanine, 2025-01-28:
The issue is that parameters are sorted alphabetically and numbers do not maintain consistent width, we would have to rename all parameters where there are 10 or more instances to always use two digits numbers. IMHO it would be nicer if we could show instantiated parameters as for example SERVO**NN**_*so we don't have to scroll or search through 32 instances of servo parameters.

Ryanf55, 2025-01-28:
We discussed in dev call yesterday and the recommended approach is to use a regex to discover numerics. We need to sort by the numerically instead of lexographically, which should solve the less-than-10 more-than-10 issue that the code currently has.

LupusTheCanine, 2025-01-28:
> We discussed in dev call yesterday and the recommended approach is to use a regex to discover numerics. We need to sort by the numerically instead of lexographically, which should solve the less-than-10 more-than-10 issue that the code currently has.

Please consider grouping them, this will make searching through parameters so much faster.

Ryanf55, 2025-01-29:
I'll see what I can do!

Ryanf55, 2025-10-29:
This issue is back open for another contributor. @shbhmexe  is wasting our time.

neo-0007, 2025-10-31:
Hi @Ryanf55 , I was looking at this issue ,
According to me the line below is causing the issue 
https://github.com/ArduPilot/ardupilot/blob/5ea5f6b245fe38d8c48be8656b3df8c0eccff958/Tools/autotest/param_metadata/param_parse.py#L443

I tried a approach by using a natural sort key (using regex to split alphanumeric names):

```python
def natural_sort_key(s):
    """Natural sort key for sorting alphanumeric strings"""
    parts = re.split(r'(\d+)', s)
    
    key = []
    for part in parts:
        if part.isdigit():
            key.append(int(part))
        else:
            key.append(part.lower())

    return key

# sort libraries by name
alllibs = sorted(alllibs, key=lambda x: natural_sort_key(x.name))
```

This seems to do the job , but the change is happening for all the parameters with alphanumeric names for example MAV1, MAV2 etc. which i think is fine .

If this approach is correct , can you assign this issue to me ?

closed 2025-11-01 by peterbarker