ArduPilot/ardupilot #29231: [BUG] Plane nav_pitch is PTCH_LIM_MAX_DEG fixed in altitude control modes (Ardupilot 4.6)

moreba1, 2025-02-05:
**Issue details**

In SITL a tailsitter VTOL , in FBWA mode it is good in control pitch but when i change the mode to Loiter or other altitude control modes, nav_pitch reach to PTCH_LIM_MAX_DEG and maintain constant and plane will climb . 

**Version**
Ardupilot 4.6

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Tailsitter

**Hardware type**
SITL

IamPete1, 2025-02-06:
What command are you launching SITL with? Where are your parameters from?

moreba1, 2025-02-06:
I used this command for SITL with Matlab Simulink.
 ../Tools/autotest/sim_vehicle.py -f JSON:127.0.0.1  --console
with these parameters:
[param.txt](https://github.com/user-attachments/files/18694449/param.txt)

[image]

IamPete1, 2025-02-07:
I suspect there is a issue with your MATLAB model, please try a known good simulator model such as realflight, https://github.com/ArduPilot/SITL_Models/tree/master/RealFlight/Released_Models/QuadPlanes/Tailsitters/SkyCat_TVBS

moreba1, 2025-02-09:
> I suspect there is a issue with your MATLAB model, please try a known good simulator model such as realflight, https://github.com/ArduPilot/SITL_Models/tree/master/RealFlight/Released_Models/QuadPlanes/Tailsitters/SkyCat_TVBS

pitch calculated correct by model and it worked before with previous Ardupilot code , 
and nav_pitch calculation related to Ardupilot code and parameters not model.

IamPete1, 2025-02-09:
> pitch calculated correct by model and it worked before with previous Ardupilot code ,
> and nav_pitch calculation related to Ardupilot code and parameters not model.

Which version of ArduPilot did it work correctly with?

moreba1, 2025-02-10:
> > pitch calculated correct by model and it worked before with previous Ardupilot code ,
> > and nav_pitch calculation related to Ardupilot code and parameters not model.
> 
> Which version of ArduPilot did it work correctly with?

Ardplane 4.5 , I recenlty found that this problem related to PTCH_TRIM_DEG , When i increase it from 0 to 4 for tail-sitter this problem occurs.

moreba1, 2025-02-12:
> known goo

IamPete1, 2025-02-12:
If DCM or the EKF does not agree with the truth values from the simulator then your not generating your simulated sensor data in a kinetically consistent way. Using `AHRS_EKF_TYPE` 10 bypasses the sensor data and directly uses the provided attitude. 

I would not have expected there to be any difference between ArduPilot versions in this area.

moreba1, 2025-02-12:
> If DCM or the EKF does not agree with the truth values from the simulator then your not generating your simulated sensor data in a kinetically consistent way. Using `AHRS_EKF_TYPE` 10 bypasses the [...]

How can i change code to TECS use Simulator attitude and height? Currently with `AHRS_EKF_TYPE` 10  nav_pitch is wrong.

IamPete1, 2025-02-12:
TECS will use the configured AHRS type.

Sorry, I'm really struggling to understand what issue your having, Please provide as many details as you can. details of your simulator, what testing has been done in other simulators, what changes you saw between versions.

moreba1, 2025-02-13:
> TECS will use the configured AHRS type.
> 
> Sorry, I'm really struggling to understand what issue your having, Please provide as many details as you can. details of your simulator, what testing has been done in other simulators, what changes you saw between versions.

I solved this issue . I must enable 'ARSPD_USE' 1 .

closed 2025-02-13 by moreba1