ArduPilot/ardupilot #29262: Quadplane set heading during QRTL mode

moreba1, 2025-02-09:
## Feature request

**Is your feature request related to a problem? Please describe.**

VTOL plane heading  change very rapidly in QRTL mode during RTL.

**Describe the solution you'd like**

I need a parameter that VTOL don't change the heading and QRTL by current heading. (or into wind)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

IamPete1, 2025-02-09:
Do you have a log showing the behavior you would like to fix?

moreba1, 2025-02-13:
> Do you have a log showing the behavior you would like to fix?

[log.zip](https://github.com/user-attachments/files/18790046/log.zip)

moreba1, 2025-02-13:
> Do you have a log showing the behavior you would like to fix?

[log.zip](https://github.com/user-attachments/files/18790161/log.zip)

I want to heading of drone don't change during QRTL but it always change heading to home during QRTL.

[image]

IamPete1, 2025-02-14:
Looks like its yawing towards the direction of travel. That is what I would have expected if it has to translate home. You would prefer it to keep current heading if its are pure VTOL reposition?

Yaw is often a issue on quad planes, there are a few ways you can reduce the yaw output. Lower gains or accel limits. Reduce yaw head room, apply PD and I term limits.

moreba1, 2025-02-14:
> You would prefer it to keep current heading if its are pure VTOL reposition?

I prefer to keep Current heading and return to home because of Wind Impact in changing yaw is dangerous for tailsitter.