ArduPilot/ardupilot #29298: SITL: WP failure in Guided mode wo GPS & EKF Origin

pavloblindnology, 2025-02-14:
## Bug report

**Issue details**

While on ground, disarmed and in Guided mode, setting global destination point (FlyTo in MavProxy) before setting EKF Origin triggers a Panic freeze in SITL, though command is rejected.
The problem happens when trying to send `POSITION_TARGET_GLOBAL_INT` message:
https://github.com/ArduPilot/ardupilot/blob/f8d13d34d9b0c5019c65a94c6ec0efe1aa8f555d/ArduCopter/GCS_Mavlink.cpp#L119-L129
which tries to change altitude frame for (0,0,0) location
https://github.com/ArduPilot/ardupilot/blob/f8d13d34d9b0c5019c65a94c6ec0efe1aa8f555d/libraries/AP_Common/Location.cpp#L109-L115
And `get_wp` in Guided with positional control (`GUID_OPTIONS` bit-6 = 0) returns True even for (0,0,0) Location.
https://github.com/ArduPilot/ardupilot/blob/f8d13d34d9b0c5019c65a94c6ec0efe1aa8f555d/ArduCopter/mode_guided.cpp#L403-L411

**Version**
Copter-4.4/4.5.

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
SITL

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2025-02-15:
Hi @pavloblindnology,

Could you test with master?  It would be good to know if we've already resolved this in the master branch before we consider going back to older releases

pavloblindnology, 2025-02-15:
Hi @rmackay9 
You're right, in master it's fixed with this commit fcebd0c8c2f13fd6cb1aa397fea7d94245851277.
What is not fixed, is that if Guided uses waypoint navigation (`GUID_OPTIONS` bit-6 = 1) then submode still changes before trying to set target. Though this doesn't lead to Panic freeze, as WP nav checks for EKF origin.

pavloblindnology, 2025-02-15:
@rmackay9 
Also please note, that arming in Guided, or switching to Guided in flight is possible wo EKF Origin set if in relative aiding (OptFlow, `ARMING_CHECK.GPS lock=0`).
With above fix from `master` that's it, but without (`Copter-4.5`) switching to Guided in flight and then trying to FlyTo leads to flying down (to (0,0,0) destination).
So, seems like additional check for EKF origin is needed when arming or when switching modes.

peterbarker, 2025-05-09:
@pavloblindnology there are various submodes in guided which don't even require an EKF origin be set - SubMode::Angle, for example.  So I don't think we can institute blanket checks as you suggest.

We should probably change things around so that we can fail to enter submodes within guided based on what condition the vehicle is in.  So, for example, to enter wpnav mode you'd need to be able to get an absolute position from the AHRS.  To enter VelAccel mode you would need to be able to get a velocity from the AHRS (eg. via OpticalFlow).  Local position commands would require EKF origin be set to be accepted.

The advantage of this is that we can remove the requirement of disabling GPS checks when you want to work with OpticalFlow.