ArduPilot/ardupilot #29306: Add support for EXTENDED_SYS_STATE

DonLakeFlyer, 2025-02-16:
QGC uses EXTENDED_SYS_STATE to understand whether the vehicle is flying, was flying or landing. This in turn lights up various QGC actions at appropriate times. With ArduPilot I can only determine some state from `HEARTBEAT.system_status` which isn't as detailed. You can see an Issue here around this: https://github.com/mavlink/qgroundcontrol/issues/12417. Would be better to support this other message tog et better flight status to QGC.

IamPete1, 2025-02-16:
`EXTENDED_SYS_STATE` is supported but it is not streamed by default, you will have to request it with the `SET_MESSAGE_INTERVAL` command.

olliw42, 2025-02-16:
https://github.com/ArduPilot/ardupilot/issues/21615

peterbarker, 2025-05-02:
@DonLakeFlyer does QGC have infrastructure for probing to see if a sysid supports a particular message type by setting its stream rate?

Would a change to support that sort of thing be an acceptable patch to QGC if it can't do this already?  Having things "on demand" allows for us to have a richer set of messages available while not overwhelming the thin pipes we're likely to have for the foreseeable future!

This is where MAVProxy probes a source system to see if it can supply the `FLIGHT_INFORMATION` message: https://github.com/ardupilot/MAVProxy/blob//MAVProxy/modules/mavproxy_console.py#L764