ArduPilot/ardupilot #29315: Boot loop after scripting enabling.

Paku-, 2025-02-17:
## Bug report

**Issue details**

Boot loop after enabling sctipts with SCR_ENABLE=1

https://discuss.ardupilot.org/t/reboot-loop-after-enabling-scr-enable-lua-scripts/130420

**Version**
4.6 beta

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
any

**Hardware type**
MicoAir743 board

**Logs**
Logs are almost empty just few first lines

Paku-, 2025-02-17:
I added th PPM on one of Serial pins, it's 4.6beta fork I play with.

https://github.com/Paku-/ardupilot/blob/paku-4.6-add-ons/libraries/AP_HAL_ChibiOS/hwdef/MicoAir743-PPM/hwdef.dat

I changed pin used for PPM twice for test but it did not help. It appears to lack any correlation.

rmackay9, 2025-02-17:
Let's keep following up on this in the forums and if we confirm that it's a real issue we can re-open this.

I'm going to close this issue, sorry to be strict but our issues list is really our to-do list where we track known bugs and enhancement requests

closed 2025-02-17 by rmackay9