ArduPilot/ardupilot #29348: V4.6.0ß2 - Obvious problems with Passthrough and AM32 ESC

robustini, 2025-02-21:
Before opening this issue, I consulted with Alka, the AM32 developer, and we cross-tested different FCs and ESCs.
From my tests, I ascertained that Andy solved the problem of incorrect requests with PR #29056 but unfortunately the flash of the ESC firmware does not work, it randomly stops, often around 50% of the process.
In the same hardware I have installed INAV V8 and have repeatedly reflashed the firmware without the slightest hesitation, always successful the first time.
So I think the passthrough on this thing continues to have unresolved problems.
My bech tests:

FC: MATEKH743
ESC: SEQURE G071 target
Ardupilot: Plane V4.6.0 ß4

I have not been able to carry out tests with other ESCs and BLHeli32 because obviously firmware flashing is no longer possible.

AlkaMotors, 2025-02-21:
The exact same issues happens flashing a blheli_s esc and esc-configurator.  The fc hangs and reboots most of the time when flashing.

rmackay9, 2025-02-22:
Just a side note that this is the PWM pass through method of updating the ESC firmware?  The **DroneCAN** method does work and it is [documented here on the AP wiki](https://ardupilot.org/copter/docs/common-am32-escs.html#esc-firmware-updates).

robustini, 2025-02-22:
> Just a side note that this is the PWM pass through method of updating the ESC firmware? The **DroneCAN** method does work and it is [documented here on the AP wiki](https://ardupilot.org/copter/docs/common-am32-escs.html#esc-firmware-updates).

Thanks Randy for the report, ESC is not meant here without DroneCAN support.

MrX1-tech, 2025-02-27:
https://github.com/ArduPilot/ardupilot/issues/29348

dmokel, 2025-03-08:
Hi, I also encountered this problem. I made a DShot AM32 ESC, which does not introduce DroneCAN, I can easily use betaflight's pass through to read and write the configuration of the AM32 ESC, but after I configured the Ardupilot parameters according to the instructions of the Ardupilot Wiki, I still cannot read and write the parameters of the AM32 ESC through AM32.ca, ESC_Config_Tool, BLHeliSuite32_32.10.0.0, and ESC Configurator. So I also think that Ardupilot still has problems with th support for AM32 Pass through.

rmackay9, 2025-03-10:
@dmokel @robustini,

Thanks for he report, I've added this to the 4.6 issues list (see link above)

boa-pe, 2025-03-24:
Today, using fresh build from commit :

`commit 1afb8d7859a884e7b2b5bcd62ebb7a66fc3bac90 (HEAD -> ArduPilot-4.6, origin/ArduPilot-4.6)`, target: MatekH743-bdshot

on Matek H743-SLIM and Skystars AM60 II web config tool works fine with pass-through - updates configuration and ESC firmware update 2.16 => 2.17 without any isues without CAN.

robustini, 2025-03-25:
> Today, using fresh build from commit :
>
> `commit 1afb8d7859a884e7b2b5bcd62ebb7a66fc3bac90 (HEAD -> ArduPilot-4.6, origin/ArduPilot-4.6)`, target: MatekH743-bdshot
>
> on Matek H743-SLIM and Skystars [...]

Thanks for the feedback but this is an isolated case, the problem is known and there is an attempted [fix](https://github.com/ArduPilot/ardupilot/pull/29590) in the works.

robustini, 2025-03-31:
Fixed with #29590.

closed 2025-03-31 by robustini