ArduPilot/ardupilot #29384: EKF3 velocity divergence pre-arm

tridge, 2025-02-27:
We had a divergence of EKF3 velocity pre-arming on the ground which caused a roll over on takeoff of a quadplane.
GPS velocity looks fine, EKF3 velocity is very bad. Innovations were low.
Replay log here:
http://uav.tridgell.net/tmp/00000082.BIN
firmware is 4.6.0-beta4, but 4.5.7 replay shows the same issue.
replays nicely, no log packet loss

[image]

[image]

tridge, 2025-02-27:
V09 is diverging:

[image]

tridge, 2025-02-27:
the V05 (variance of velocity east) is diverging badly:
[image]
it is stopping at 1000.0, which comes from NavEKF3_core::ConstrainVariances()

priseborough, 2025-03-02:
Replay of the log with EK3_MAG_CAL=4 (always use 3-axis fusion) eliminates the instability so I am now investigating the yaw fusion process.

timtuxworth, 2025-03-03:
I don't know if this is related, but I had a strange rollover on takeoff on a Fixed Wing yesterday. One thing I noticed is that it was pointing almost directly North just before the takeoff.  

https://www.dropbox.com/scl/fi/grw2cn5ru0329j50gzck7/log_13_2025-3-2-18-10-36.bin?rlkey=xz8u3f6y64nun5ui4hu06euiy&dl=0 

Sorry if this is not related.

rmackay9, 2025-08-21:
Hi there, I've added "wikiNeeded" to this because we should add it to this wiki page https://ardupilot.org/copter/docs/common-prearm-safety-checks.html.

To justify adding it to the wiki page, here's a user who has bumped into the arming check: https://discuss.ardupilot.org/t/error-message-ekf3-yaw-inconsistent-by-62-deg/80647/8

tpwrules, 2026-09-13:
Fix has been merged and AFAIK we've seen no new reports.

closed 2026-09-13 by tpwrules