ArduPilot/ardupilot #29392: Setting TERRAIN_SPACING Too Small Causes Instant Crash Instead of Running Out of Storage

wyunc, 2025-02-27:
## Bug report

**Issue details**
Hi~
When the parameter TERRAIN_SPACING is set to a value less than 1, the system crashes, causing the GCS to lose connection.
For example, param set TERRAIN_SPACING 0 or param set TERRAIN_SPACING 0.5
(But there are no errors when TERRAIN_SPACING =1 or 1.5)

According to the parameter [description](https://ardupilot.org/copter/docs/parameters.html#terrain-spacing-terrain-grid-spacing):
"Grid spacings lower than 100 meters waste SD card space if the GCS cannot provide that resolution."

If the value is set below 100, it should only result in wasted storage space (eventually leading to an SD card running out of space), but it should not cause an instant crash. Therefore, I think this is a bug.

The error message from debugging is as follows:
Thread 1 "arducopter" received signal SIGFPE, Arithmetic exception.
0x00005555556979a4 in AP_Terrain::calculate_grid_info(Location const&, AP_Terrain::grid_info&) const ()

**Version**
SITL, copter, 4.7.0-dev

**Platform**
[  ] All
[  ] AntennaTracker
[*] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

peterbarker, 2025-02-27:
This is an integer parameter.  It's being truncated to zero, leading to division-by-zero.

You *really* want sub-metre accuracy?

rmackay9, 2025-02-27:
Maybe we should probably protect against unrealistically small values especially if they can cause problems on a real vehicle

wyunc, 2025-02-28:
Thank you @peterbarker  and @rmackay9 ,
In my opinion, if you want to fly indoors (within a 50-square-meter area), setting the grid resolution to sub-meter might be a reasonable approach. I tend to agree with what @rmackay9  suggested — it would be best to prevent this situation from happening.

I think the simplest solution would be to add a constraint in the code, so that when the grid resolution is set to less than 1, it defaults to 1. Alternatively, we could add a note in the documentation stating: "Do not set TERRAIN_SPACING to less than 1."

Another option could be to enhance support for sub-meter resolution, though I'm not entirely sure if that's necessary.

PR #29653 (2025-05-14): AP_Terrain: Protect against division by 0 when TERRAIN_SPACING is less then 1
This fixes #29392. In the calculate_grid_info function and the east_blocks function of AP_Terrain, the TERRAIN_SPACING parameter is used to determine grid spacing and how many blocks are needed in a stride for a given location. 

When this parameter is set to a value less than 1, a division by zero occurs, leading to an arithmetic exception. This fix   ensures that grid_spacing is properly handled when its value is below 1. Specifically, if grid_spacing is less than 1, the system defaults to using a value of 1 instea [... cut]
closed 2025-05-14 by peterbarker