ArduPilot/ardupilot #29408: Is There an SDK Library for ArduPilot Similar to MAVSDK?

Bhuvnesh-codeboosters-internship, 2025-03-01:
I have previously used the MAVSDK library (GitHub link: http://github.com/mavlink/MAVSDK) to connect to PX4, and it worked well. However, we would like to move to ArduPilot due to a customer requirement. Therefore, I am looking for an alternative SDK that can effectively work with ArduPilot.
 
Are there any SDK libraries available in Python or Java that are similar to MAVSDK and compatible with ArduPilot? Please let me know if there are any suitable options available.

rmackay9, 2025-03-01:
Hi @BhuvneshM008,

Thanks very much for the question.  In a general sense MAVSDK should also work with AP it's just that not a lot of development team effort has been put into resolving any issues that may arise.  Have you tried MAVSDK with AP and found issues?

At a lower level, pymavlink can be used for the mavlink communication and we have [details of AP's mavlink interface here on the wiki](https://ardupilot.org/dev/docs/mavlink-commands.html).

peterbarker, 2025-04-29:
MAVSDK can be used, FWIU.  dronekit-python, while essentially abandoned, is still surprisingly popular.  Raw pymavlink, as @rmackay9 suggests.

closed 2025-04-29 by peterbarker