ArduPilot/ardupilot #29416: Rover: add support for CONDITION_YAW mission command

rmackay9, 2025-03-03:
Skid steering rovers and boats have the ability to control their heading so we could add support for the CONDITION_YAW command while in Auto and Guided mode.  This would allow users to more easily point their vehicle in a specified direction while they're not moving.

jrcacador, 2025-04-28:
Hi! Following the suggestion on [discord](https://discord.com/channels/674039678562861068/674039678982422579/1364797839309537320), I'm currently working on this issue

rmackay9, 2025-04-29:
One tricky thing about handling this issue while the vehicle is in Auto mode is that condition-yaw is a "do" command instead of a navigation command.  This means that the command will likely need to be placed after a waypoint command that includes a long delay.  For guided mode this isn't an issue.