ArduPilot/ardupilot #29423: AP_DDS: 1 meter Lat-lon resolution

Ryanf55, 2025-03-04:
## Bug report

**Issue details**

Loss of precision from float default init.

https://github.com/ArduPilot/ardupilot/pull/28449#discussion_r1978410454

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

tizianofiorenzani, 2025-03-04:
Addressed in https://github.com/ArduPilot/ardupilot/pull/28449

Kalyani02072003, 2025-03-22:
Hi @Ryanf55, I'd like to contribute to this issue. Is it still open for contributions? Let me know how I can help!

Ryanf55, 2025-03-22:
> Hi [@Ryanf55](https://github.com/Ryanf55), I'd like to contribute to this issue. Is it still open for contributions? Let me know how I can help!

No, sorry.  see one comment above. Tiziano has a solution.

sanggusti, 2025-03-31:
Hi, since this issue is already resolved by #28449 I think this issue could be closed.

closed 2025-03-31 by Ryanf55