ArduPilot/ardupilot #29424: AP_DDS: Publish at least once when health fails

Ryanf55, 2025-03-04:
## Bug report

**Issue details**

If GPS was healthy then goes unhealthy, a subscriber won't know because we don't publish unhealthy data ever. We should publish this once for reliable transport, or at a low rate if unrelative transport
_Please describe the problem_

**Version**
master
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

abdulrahman-1212, 2025-03-14:
Hi @Ryanf55 ,
I am really interested to work on this issue but I need some guidance from you since this is my first open source contrib. As I understand I am supposed to check the state of the GPS and if unhealthy, send some feedback to subscribers at least once.

Ryanf55, 2025-03-14:
> Hi [@Ryanf55](https://github.com/Ryanf55) , I am really interested to work on this issue but I need some guidance from you since this is my first open source contrib. As I understand I am supposed to check the state of the GPS and if unhealthy, send some feedback to subscribers at least once.

Yea. I'll put together an example for one channel, and you can copy it to the rest.