ArduPilot/ardupilot #29442: FS_CRASH_CHECK Pitch and Roll

mhornsby, 2025-03-05:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REQUEST WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please do a careful search before opening this, there are already a lot of feature requests**_
If there was a previous discussion in the forum, link to it

## Feature request

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

Wondering if [FS_CRASH_CHECK](https://ardupilot.org/rover/docs/parameters.html#fs-crash-check) can be expanded to include an option for pitch and roll. My Rover is running on mild terrian so if say Pitch or Roll is more than +/- 45 then its likely the rover is in trouble e.g. run a wheel up fence or tree

**Additional context**
_Add any other context or screenshots about the feature request here._

Zandriy, 2025-05-21:
Looking from the code of Rover::crash_check() I see  the roll and pitch angles are covered already by [FS_CRASH_CHECK](https://ardupilot.org/rover/docs/parameters.html#fs-crash-check), but you should set another parameter for point the allowed roll and pitch angles [CRASH_ANGLE](https://ardupilot.org/rover/docs/parameters.html#crash-angle).
So I think to solve the problem in the described case two parameters should be set:
FS_CRASH_CHECK = FS_CRASH_HOLD_AND_DISARM
CRASH_ANGLE = 45

Let me know if I misunderstood the problem.

rmackay9, 2025-05-21:
Yes @Zandriy is correct!  This apparently went in as part of our [balance bot support](https://github.com/ArduPilot/ardupilot/pull/8701) but we never documented it on the wiki!  I've created an issue for the wiki here https://github.com/ArduPilot/ardupilot_wiki/issues/6880

closed 2025-05-21 by rmackay9