ArduPilot/ardupilot #29459: Quadplane Servo_out PWM set incorrectly to Q_M_PWM_MIN instead of SERVO_MIN in Disarm

moreba1, 2025-03-07:
**Issue details**

Hello

I will set Q_M_PWM_MIN  to limit minimum throttle of quadplane in QLOITER and QHOVER mode in flight , but when the drone is disarmed it is also Q_M_PWM_MIN value  and is i will it set to SERVO_MIN value .

**Version**
plane 4.5.7

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

moreba1, 2025-03-07:
when i set Q_M_SAFE_DISARM to 1 , it set 0 PWM instead of Servo_out PWM.

moreba1, 2025-03-07:
> That is the correct behaviour. Q_M_SPIN_MIN should be used for seting a lower limit in flight.

yes it is correct in flight but i will it set to SERVO?_MIN while it is disarmed.

moreba1, 2025-03-07:
> > That is the correct behaviour. Q_M_SPIN_MIN should be used for seting a lower limit in flight.
> 
> yes it is correct in flight but i want it set to SERVO?_MIN while drone is disarmed.

moreba1, 2025-03-07:
> That is not possible.

why? when the drone is DisArmed on ground i want to set 1000 pwm (not 0) , minimum pwm value that ESC need is 1000. 
if i set Q_M_SAFE_DISARM  to 1 it set 0 pwm and else it set to Q_M_SPIN_MIN that is more than 1000 and while it is DisArmed .

moreba1, 2025-03-07:
> > when the drone is DisArmed on ground i want to set 1000 pwm (not 0)
> 
> Set `Q_M_PWM_MIN` to 1000.
> 
> Please explain in detail why that does not work in your use case.

I need a minimum Thrust in hover, so i can not set it to 1000, i need a minimum 1500 pwm in Flight and 1000 pwm in disarm.

IamPete1, 2025-03-08:
> I need a minimum Thrust in hover, so i can not set it to 1000, i need a minimum 1500 pwm in Flight and 1000 pwm in disarm.

I don't think what your asking for makes sense, you will get a pre-arm, but you can do it like this:

`Q_M_PWM_MIN` 1000
`Q_M_PWM_MAX` 2000
`Q_M_SPIN_MIN` 0.5

If you can explain what your actually trying to achieve (and why) we may be able to suggest a alternate approach.

moreba1, 2025-03-12:
> I don't think what your asking for makes sense, you will get a pre-arm, but you can do it like this:
>
> `Q_M_PWM_MIN` 1000 `Q_M_PWM_MAX` 2000 `Q_M_SPIN_MIN` 0.5
>
> If you can explain what your actually [...]

Yes , it worked thanks.

closed 2025-03-12 by moreba1