ArduPilot/ardupilot #29476: Problem with servos and B\E led FMU on Pixhawk 6c mini

sartor, 2025-03-09:
## Bug report

I have brand new pixhawk 6c mini. After power on (by any source) B\E led in FMU frame blinking twice repeatedly. And all servo outputs not working (I checked it with different servos and oscilloscope).
For debug purpose I’ve flashed PX4 and B\E led blinking and servos works fine.
I’ve tried 4.5.7 and 4.3.8 versions. No difference. Bootloader is updated. I’ve read similar issues, but no solution found.

**Version**
I have tried Plane 4.5.7, Plane 4.6.0 beta and Plane 4.3.8

**Platform**
Plane 

**Hardware type**
Pixhawk 6c mini

Here is video of LED status after power on:
https://youtu.be/AvKIcysHon0

IamPete1, 2025-03-09:
Have you configured the safety switch?

https://ardupilot.org/plane/docs/common-safety-switch-pixhawk.html

rmackay9, 2025-03-10:
This sounds like a support issue so it should go into the support forum https://discuss.ardupilot.org/c/arduplane/7

I'm going to close this issue now but we will see replies.

Sorry to be strict but if we're not then our to-do list (which is what this is) turns into another support forum.

Thanks for using AP and reporting back on issues.

closed 2025-03-10 by rmackay9