ArduPilot/ardupilot #29500: Custom Build Server failure if everything disabled

rmackay9, 2025-03-11:
While I noticed this using the custom build server, I suspect it's an AP flight code include issue

../build_src/libraries/AP_Compass/AP_Compass.cpp: In member function 'void Compass::probe_i2c_spi_compasses()':
./hwdef.h:683:37: error: 'DRIVER_LSM303D' was not declared in this scope
  683 | #define HAL_MAG_PROBE1  ADD_BACKEND(DRIVER_LSM303D, AP_Compass_LSM303D::probe(hal.spi->get_device("lsm9ds0_ext_am"),ROTATION_YAW_270))
      |                                     ^~~~~~~~~~~~~~
../build_src/libraries/AP_Compass/AP_Compass.cpp:1100:30: note: in definition of macro 'ADD_BACKEND'
 1100 |     do { if (_driver_enabled(driver_type)) { _add_backend(backend); } \
      |                              ^~~~~~~~~~~
./hwdef.h:687:28: note: in expansion of macro 'HAL_MAG_PROBE1'
  687 | #define HAL_MAG_PROBE_LIST HAL_MAG_PROBE1;HAL_MAG_PROBE2;HAL_MAG_PROBE3;HAL_MAG_PROBE4
      |                            ^~~~~~~~~~~~~~
../build_src/libraries/AP_Compass/AP_Compass.cpp:1412:5: note: in expansion of macro 'HAL_MAG_PROBE_LIST'
 1412 |     HAL_MAG_PROBE_LIST;
      |     ^~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.

peterbarker, 2025-04-24:
@rmackay9 this comes about because the hwdef is saying, "there is a sensor of this type on this bus at this address".  So when we create the hwdef.h for the board we add lines saying, "probe using this sensor driver backend at this bus and address".

We do not check that that sensor driver backend is actually compiled into the code - we just attempt to call the method.

We could relatively easily wrap the call to the probe method in `#ifdef`s.  We have not done that to date because it is considered unlikely that the user would *not* want to compile in support for the compasses which are physcially soldered to their autopilot's main board.  So instead we throw ugly compiler errors at them.  We have special carve-outs in various sanity-checking tools because we don't have these #ifdefs (https://github.com/ardupilot/ardupilot/blob/master/Tools/autotest/test_build_options.py#L79)

I would be perfectly happy with the `#ifdef`s to avoid the compilation failures.  A user could have a legitimate reason for not wanting support on their boards.

rmackay9, 2025-04-24:
Hi @peterbarker,

OK, thanks for the feedback.  I just think that we need to catch the issue earlier somehow.  If possible, let's not give the user options that are not actually going to compile.  Maybe that means greyed out checkboxes that can't be unchecked, maybe something else, I'm not really sure