ArduPilot/ardupilot #29515: LittleFS: Unknown SPI Flash 0x00000000

be-ska, 2025-03-13:
## Bug report

**Issue details**

Using a custom hardware with W25N01GV flash chip, the LittleFS filesystem _sometimes_ can't be mounted, MAVProxy report:

```
Unknown SPI Flash 0x00000000
FlashMemory_LittleFS: dead
SDCard failed to start
```
The same hardware works flawlessly with the previous batch logging functionality.

I could fix the issue editing `mount_filesystem()`, calling `init_flash()` **before** `find_block_size_and_count()`.

https://github.com/ArduPilot/ardupilot/blob/d4389d1279d72287d94be0515f23ecac7b268959/libraries/AP_Filesystem/AP_Filesystem_FlashMemory_LittleFS.cpp#L879-L891

It actually seems reasonable to reset and init the flash chip before requesting the JEDEC ID, I could open a PR with this simple modification if needed.

**Version**
4.7.0-dev

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Custom hardware with H753 and W25N01GV SPI flash chip

tpwrules, 2025-03-13:
This seems reasonable, please send a PR.

peterbarker, 2025-05-02:
OK, so this is an acknowledged problem, but the first PR didn't work out.  @be-ska do you have another idea yet?

be-ska, 2025-05-05:
This has been a weird issue to debug. Out of the five flight controllers I tested with LittleFS, only one exhibited this problem, and it was a recurring issue. Using the patch in the first pull request the issue seemed to be resolved, but after some flights (I’m not sure exactly how many, since I can’t even read the logs), the problem reappeared.
I also tried calling init_flash() in a loop for 10 seconds or until it returned true, and added some delays. This seemed to help at first, but after a while, the issue came back. I stopped testing at this point because I need to focus on other projects.
It’s worth noting that with the old but gold batch logging method I never experienced any failures. My idea was to compare the two implementations to identify any differences, but unfortunately I don’t have the time to do that at the moment.