ArduPilot/ardupilot #29520: Quad Plane vectored Tilt Tri - FBW Takeoff Problem

axelisart, 2025-03-13:
## Bug report

**Issue details**

Quadplane vectored tilt tri does not allow FBW takeoff despite having set Q_OPTIONS bit for it (Allow FW Takeoff). Arduplane does not take care of any of the TKOFF options like TKOFF_ACCEL_CNT or TKOFF_THR_MAX just to name few. Instead, the controller push all the motors to 100% instantaneously which could lead to a disaster...

**Version**
4.5.7

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadplane of type vectored Tilt TRI

**Hardware type**
Pixhawk 6C Mini

**Logs**
It is easy to reproduce on bench so I can do more tests if you want.

[2025-03-13 17-40-31-anon.zip](https://github.com/user-attachments/files/19234034/2025-03-13.17-40-31-anon.zip)

raul-ortega, 2025-04-10:
Hi! This is has been an important issue which I have been fighting to in my latest sixth flights. I have try with several values to TKOFF_THR_MAX (95% ... 65%, 55%) to try to reduce high power consumption in combination with BATT_WATT_MAX and Q_M_BAT_CURR_MAX, without success. In all cases consuption is always the same, very high during transition from Q_LOITER to FBWA mode.

In the last flight I tried with TKOFF_THR_MAX set to 45% and the result was that the plane moved backwards due to the wind. The servos didn't tilt the motors forward; instead, they remained completely vertical. I didn't use the throttle stick, I never use it when transitioning from Q_LOITER to FBWA. The motors have a certain amount of power, and at 50% throttle, they take off like a rocket. It would be nice to have a variable that would allow a specific throttle percentage to be applied during the transition, or TKOFF_THR_MAX, which would allow the tilt to be lowered. This would reduce power spikes and keep the battery safe.

As an alternative I'll try to reduce THR_MAX for may next flight, but this is something that I don't like too much because it might cause certain inconvenience in som situations in which would be needed increasing throttle from the stick.

raul-ortega, 2025-04-10:
I have uploaded to log, one with TKOFF_THR_MAX set to 55%, and the log off my last attempt set to 45%

[log_TKOFF_THR_MAX_55.zip](https://github.com/user-attachments/files/19683198/log_TKOFF_THR_MAX_55.zip)

[log_TKOFF_THR_MAX_45.zip](https://github.com/user-attachments/files/19683199/log_TKOFF_THR_MAX_45.zip)