ArduPilot/ardupilot #29530: AP_Camera_Backend::Write_CameraInfo does not handle uninitialized location

blaab2, 2025-03-14:
## Bug report

**Issue details**

This is a HAL_BOARD_SITL bug report. The panic function within _libraries/AP_Common/Location.cpp_ is triggered if a camera trigger is requested without the location being available.

**code details**

 This occurs as the 
https://github.com/ArduPilot/ardupilot/blob/7601c8388eee00650632e6b323f7101c80372002/libraries/AP_Common/Location.cpp#L109
 panic function will be raised by `AP_Camera_Backend::Write_CameraInfo` if camera trigger is requested (for example via mission planner) and location is not known yet. The unguarded call to `current_loc.get_alt_cm` raises this.

**fix**

https://github.com/ArduPilot/ardupilot/blob/7601c8388eee00650632e6b323f7101c80372002/libraries/AP_Camera/AP_Camera_Logging.cpp#L33
and
https://github.com/ArduPilot/ardupilot/blob/7601c8388eee00650632e6b323f7101c80372002/libraries/AP_Camera/AP_Camera_Logging.cpp#L37
should be
`if (camera_feedback.location.initialised() && !current_loc.get_alt_cm(Location::AltFrame::ABSOLUTE, altitude_cm)) {`
and
`if (camera_feedback.location.initialised() && !current_loc.get_alt_cm(Location::AltFrame::ABOVE_HOME, altitude_rel_cm)) {`

**Version**
master branch

**Platform**
[?] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_irrelevant_

**Hardware type**
_irrelevant_

**Logs**
```
#5  0x000055ebb20e1175 in AP_HAL::panic (errormsg=0x55ebb2231600 "Should not be called on invalid location: Location cannot be (0, 0, 0)") at ../../libraries/AP_HAL_SITL/system.cpp:51
        ap = {{gp_offset = 8, fp_offset = 48, overflow_arg_area = 0x7ffcabb23be0, reg_save_area = 0x7ffcabb23b20}}
```

rmackay9, 2025-03-18:
Hi @blaab2,

Thanks very much for the report.  [I've created a PR to resolve the issue](https://github.com/ArduPilot/ardupilot/pull/29558) and I've "Co-Authored" it with you to give you the credit because you've done most of the work.  Please tell me if it looks OK to you.

Thanks very much!

rmackay9, 2025-03-18:
This has been merged to master and should go out with -beta5 in a few days, thanks!

closed 2025-03-18 by rmackay9