ArduPilot/ardupilot #29545: Handling of sensor reconfiguration in ArduPilot

axoaxel, 2025-03-17:
## Bug report

**Issue details**

Our recent research has identified an implicit trust in sensors connected to the flight controller. Upon sensor reconfiguration, Ardupilot fails to detect the incorrect configuration of the peripherals. Sensor reconfiguration can destabilize UAVs, leading to crashes or significant deviations from planned trajectories, without triggering the failsafe. 

##### Steps to Reproduce:

A message is injected into the UAV's local sensor bus to reconfigure sensors (e.g., I2C, SPI), such as suspending their operation or changing the sensor operating frequency. The sensor will start behaving differently from expected. The data coming from the reconfigured sensor is processed by the UAV's flight controller, leading to incorrect state estimations. The UAV executes erroneous control actions based on these faulty estimations, resulting in destabilization or crashes.

As a POC, we provide an example for the BMI270, BMI055 sensor by changing the firmware to operate with a suspended sensor or at a lower frequency (by relying on the information provided in the sensor's datasheet). In our research, we identify that this can be done without changing the firmware.

For BMI270, 
Inside read_fifo() https://github.com/ArduPilot/ardupilot/blob/9fb1e2b9f7b746716397aa1d298943c201d9dd8a/libraries/AP_InertialSensor/AP_InertialSensor_BMI270.cpp#L374

    // Add the following for a frequency change
    write_register(BMI270_REG_ACC_CONF, 1U<<7 | 0x06); // set Accelerometer frequencyto 25Hz
    write_register(BMI270_REG_ACC_RANGE, 0x03);
    write_register(BMI270_REG_GYRO_CONF, 1U<<7 | 1<<6 | 2<<4 | 0x06); //set Gyro frequency to 25Hz
    write_register(BMI270_REG_GYRO_RANGE, 0x08);

    // Add the following for sensor suspend
    write_register(BMI270_REG_PWR_CONF, 0x01); //Advanced power mode enabled.
    hal.scheduler->delay(1);
    write_register(BMI270_REG_PWR_CTRL, 0x00); //Disable the Gyro and Accelerometer

For the BMI055, inside the read_fifo_accel() https://github.com/ArduPilot/ardupilot/blob/9fb1e2b9f7b746716397aa1d298943c201d9dd8a/libraries/AP_InertialSensor/AP_InertialSensor_BMI055.cpp#L239

    // Add the following for a frequency change
    dev_accel->write_register(REGA_PMU_BW, 0x09, true); //set frequency to 15.63 Hz

    // Add the following sensor suspend 
    dev_accel->write_register(REGA_PMU_LPW, 0x80, true); //set accelerometer suspend

inside the read_fifo_gyro() https://github.com/ArduPilot/ardupilot/blob/9fb1e2b9f7b746716397aa1d298943c201d9dd8a/libraries/AP_InertialSensor/AP_InertialSensor_BMI055.cpp#L299

    // Add the following for a frequency change
    dev_gyro->write_register(REGG_BW, 0x87, true); //set ODR 100Hz with filter bandwidth 32Hz

    // Add the following sensor suspend 
    dev_gyro->write_register(REGG_LPM1, 0x80, true); // set gyro sensor suspend

##### Expected Behavior:

The UAV should detect the sensor configuration change, trigger the failsafe, or reissue the correct configuration.

##### Actual Behavior:

The UAV fails to detect the sensor reconfiguration, processes incorrect sensor data, and performs destabilizing control actions, leading to potential crashes or significant deviations from its intended path.

Specifically, we identify three cases.

- **Armed State with Sensor Frequency Change**: No alarms are raised when a sensor's frequency is modified during flight (i.e., when the drone is armed). The drone continues to operate with the altered sensor update rate, leading to flight destabilization or crashes due to inadequate sensor data processing. From our experiments with SITL, we found that this behavior will not trigger the failsafe or the 'main loop slow' error.

- **Sensor Suspend**: Disabling a sensor during operation results in varied behaviors within the control system, depending on the sensor behavior during suspend. Specifically, we identify three possible behaviors for the sensor suspend: Default Data (the sensor, when queried, responds to the microcontroller with a constant value), Absent Data (No response to the data read), and Stale Data (the last sensed value is repeatedly reported to the controller). Those three sensor behaviors will result in two consequences at the flight controller: 
    1. **Control Loop Halting**: The control loop stalls, leading to a loss of control and potential crashes. This behavior will not trigger the failsafe.
    2. **Erroneous State Estimation**: The system might resort to using default or stale data, which can cause incorrect control responses and instability. This behavior will trigger the failsafe, but from our experiments in SITL, there is a high time to detect for such manipulations.

- **Unarmed State with Sensor Frequency Change**: Additionally we observed in SITL that altering a sensor's update frequency while the drone is unarmed triggers a 'main loop slow' error.

##### Impact:

Sensor reconfiguration can result in UAV crashes or significant deviations from planned trajectories (over 30 meters), posing safety risks and potential mission failures.

##### Recommendations:
Since current chip designs cannot prevent sensor reconfiguration, a viable solution is to periodically attest the sensor configuration. While this approach does not eliminate the underlying issue, it helps mitigate the implicit trust assumption associated with sensors.
- **Sensor Configuration Attestation**: Implement mechanisms to verify the integrity and authenticity of sensor configurations and data to detect unauthorized modifications (attestation).

**Version**
version 4.4.0-DEV pulled on January 2023, applicable to newer verisons.

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N.A.

**Hardware type**
Various hardware setups. Tested the reconfiguration of the following flight controllers i) Kakute H7 Mini (BMI270 IMU); ii) Pixhawk 6C with a dual-IMU (BMI055 IMU and ICM-42688-P IMU); iii) Omnibus F4 (MPU6000 IMU). 

**Logs**
In the logs we show the behavior observed for the BMI270 under sensor reconfiguration.

Suspend, flight controller on a table --> default data behavior, producing erroneous state estimation.
[image]

Frequency reconfiguration, flight controller on a table --> 
As long as the sensor sampling frequency is higher than 400Hz, the reconfiguration does not show any consequence on the controller. When the sensor frequency is reduced below 400Hz, we can observe that the Kalman Filter estimation gets slowed down. Moreover, we observed scheduled tasks in the ROTS being slowed down (e.g., barometer sensor will be sampled less frequently).
[image]

IamPete1, 2025-03-17:
I certainly agree that if this were to happen it would be bad, but how would this happen in the wild?

Obliviously we can't protect against someone changing the firmware to introduce such changes. I guess someone could use another device on the bus to change the configuration, but if they are going to that effort then there are lots of other things they could do too.

axoaxel, 2025-03-18:
Hi, thanks for the quick reaction to the issue.

While the concept of sensor reconfiguration is general, in our research we envision two possibilities for this reconfiguration to occur:
- Through the use of another device on the bus (as it was correctly pointed out in the previous comment).
- Through the use of intentional electromagnetic emissions (IEMI).

Abstracting from how the reconfiguration is achieved, ArduPilot is not reacting to those changes at the moment causing the described impact on the control loop (slowing down or halting the control routines).

LupusTheCanine, 2025-04-24:
@peterbarker while hostile on the bus (HoB) is a state level actor threat and we can't really mitigate that as HoB can reconfigure faster than we do or just stall/flood the bus (I2C, CAN,  network). IEM attacks will be hard to diagnose (loss of vehicle, transient failure) and likely will be underreported due to misidentification, environment not conductive to bug reporting, lack of user awareness.

Do we have CVE register?

axoaxel, 2025-04-24:
@peterbarker, thanks for pointing this out. I believe that this goes in the right direction. As I mentioned earlier, attestation would not take away the root issue, but enable detection (@LupusTheCanine comment stresses that reconfiguration can repeatedly occur).   

By inspecting the code, I realized that the checks are done synchronously within a read_fifo, but if the data is not available the function returns upon checking the registers.

axoaxel, 2025-04-24:
@peterbarker Thanks for pointing out the checked registers—this is definitely the right direction for sensor attestation. As mentioned in my previous comment and noted by @LupusTheCanine, attestation by itself doesn’t remove the root issue, but it does enable us to detect it.

While reviewing the BMI055 driver, I noticed that register checks only occur if new data arrives (if `read_registers` calls fail, the code returns without calling the `check_next_register`). One way to improve this would be to run those checks asynchronously, rather than tying them to data availability. In the current setting, upon sensor suspension, no data would be available, preventing the check from taking place.

peterbarker, 2025-05-01:
OK, it would be nice to keep these issues as actionable and closable (we have an unmanageable 1500 issues open ATM!)

This is by no means the worst issue in our repository, but, "do things really, really well" doesn't make for a good issue (more of a Wiki page, or a discussion forum topic :-) .

To that end, could we keep this tight by re-titling this issue to "Pro-actively check_registers instead of on data receipt" to make this something that can actually be ticked of as done?

Along those lines.  We don't currently check a huge number of registers.  We can't afford flash space for all of them.  It might be feasible to add more as a compile-time option for those that feel they need it.  But please note I can't see anyone on the Dev Team getting really enthusiastic about the work, so we'd be relying on contributions from outside.