ArduPilot/ardupilot #29563: Internal error from multi-heap allocation

peterbarker, 2025-03-18:
## Bug report

**Issue details**

An ArduPilot partner running a custom frame configured via scripting is getting internal errors and failure-to-allocate errors.

```
2025-03-18 12:29:53 PM : PreArm: Internal errors 0x100000 l:58 flow_of_ctrl
```

That's from this:
```
    if (heapp->current_heap_usage != 0) {
        // lua should guarantee that there is no memory still
        // allocated when we destroy the heap. Throw an error in SITL
        // if this proves not to be true
        INTERNAL_ERROR(AP_InternalError::error_t::flow_of_control);
    }
```

**Version**
stable, "latest" and 4.6.0-beta4

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
scripted

**Hardware type**
SITL

**Logs**
Ask partner

tpwrules, 2025-03-19:
We weren't sure if there was a perhaps-error-case where the error won't hold. I don't think it matters on a real vehicle but it will cause a memory leak in SITL. It might also cause a problem with reallocation which would indeed be bad.

Can the script be shared so I can reproduce locally?

tpwrules, 2025-03-21:
I am able to replicate this using mission planner SITL on real Windows by restarting scripting enough times (sometimes zero times). In my case the internal error is always paired with this error: https://github.com/ArduPilot/ardupilot/blob/d5098c6b0c3d4697f1d95fc829d1ed4089bd75f2/libraries/AP_Scripting/lua_scripts.cpp#L493-L497 which AFAICT only happens if Lua can't allocate memory to start.

This suggests the heap is not being initialized properly which also explains why it's failing to be consistent on shutdown. We sort of assume it's initialized based on our various hacks to make malloc/new zero memory blocks on allocation, but that keeps [breaking on various platforms](https://github.com/ArduPilot/ardupilot/pull/29005) and in various ways and I think it has again here. I do not personally plan to fix it again.