ArduPilot/ardupilot #29564: Failing to generate hwdef for BBBmini

uuuuuuummmmmmm, 2025-03-18:
## Bug report

**Issue details**
When trying to run ./waf configure --board=bbbmini to build the ardupilot firmware, an error occurs. This error occurs both on my physical BBBmini and in the custom firmware builder. 
It says it can't generate the hwdef and fails to configure for the BBB.

**Version**

latest 

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**

BBBmini 

**Logs**

git hash from the custom firmware builder: https://github.com/ArduPilot/ardupilot/commit/1aacdc53ba

peterbarker, 2025-04-24:
You're going to need to provide more details.

configure works for me for that board both on master and the commit you reference.

All you really needed to do was paste the text output of the error message :-)

@uuuuuuummmmmmm more information?

uuuuuuummmmmmm, 2025-04-24:
This was solved in the discord, after I brought it there when writing this issue. I simply forgot to close it.

closed 2025-04-24 by uuuuuuummmmmmm