ArduPilot/ardupilot #29566: DO_SET_ROI_LOCATION and DO_SET_ROI_NONE "gimbal device id" fields should be supported

rmackay9, 2025-03-19:
AP supports the  [DO_SET_ROI_LOCATION](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_ROI_LOCATION) and [DO_SET_ROI_NONE](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_ROI_NONE) as "immediate" and mission commands but we do not correctly consume "param1" ("Gimbal device ID").  Instead it is always the first camera gimbal that is controlled.  We should consume this field and control the gimbal device specified

rmackay9, 2025-07-06:
Hi @G1-k, thanks for looking into this.  yes, I think the do-set-roi-location and do-set-roi-none should consume the gimbal id field in the same way as do-gimbal-manage-pitchyaw handler.  thanks!

ahmedmsalah99, 2025-07-17:
Hello @rmackay9 
This is my first contribution and I have added and tested this feature. Can you please check ?

ahmedmsalah99, 2025-07-19:
@rmackay9 
I know it's a big push that's hard to review. Should we make this issue for the auto mode only ?
I can keep auto mode commits only for copter and plane which are well tested. 

For the other parts we can create separate issues to make them easier to test. 
What do you think ?

bringert, 2026-09-10:
While it would be nice to have support for specifying the gimbal device id to set the ROI for, maybe a simpler first step would be to just support param1 == 0 correctly, that is, to set the ROI for **all gimbals**, instead of only for the first one? That would be enough for my current use case at least.