ArduPilot/ardupilot #29572: Commit c0ce5e5 causes Invensensev3 INS calibration to hang forever

heeplr, 2025-03-19:
## Bug report

Since commit c0ce5e5, the ICM42688P (Invensensev3) on my board doesn't initialize anymore.

Backtrace:

```c++
...
#7  0x0000000000483e44 in AP_InertialSensor::wait_for_sample (this=0x8934c0 <rover+2960>) at ../../libraries/AP_InertialSensor/AP_InertialSensor.cpp:2128
#8  0x00000000004832b8 in AP_InertialSensor::update (this=0x8934c0 <rover+2960>) at ../../libraries/AP_InertialSensor/AP_InertialSensor.cpp:1906
#9  0x000000000048236c in AP_InertialSensor::_init_gyro (this=0x8934c0 <rover+2960>) at ../../libraries/AP_InertialSensor/AP_InertialSensor.cpp:1734
#10 0x0000000000481448 in AP_InertialSensor::init_gyro (this=0x8934c0 <rover+2960>) at ../../libraries/AP_InertialSensor/AP_InertialSensor.cpp:1421
#11 0x0000000000480974 in AP_InertialSensor::init (this=0x8934c0 <rover+2960>, loop_rate=50) at ../../libraries/AP_InertialSensor/AP_InertialSensor.cpp:954
#12 0x000000000041e700 in Rover::startup_INS (this=0x892930 <rover>) at ../../Rover/system.cpp:288
#13 0x000000000041df7c in Rover::init_ardupilot (this=0x892930 <rover>) at ../../Rover/system.cpp:123
#14 0x00000000004f1a90 in AP_Vehicle::setup (this=0x892930 <rover>) at ../../libraries/AP_Vehicle/AP_Vehicle.cpp:424
#15 0x00000000005d80fc in HAL_Linux::run (this=0x8a2318 <hal_linux>, argc=13, argv=0x7fffffffef58, callbacks=0x892930 <rover>) at ../../libraries/AP_HAL_Linux/HAL_Linux_Class.cpp:494
#16 0x000000000040cb50 in main (argc=13, argv=0x7fffffffef58) at ../../Rover/Rover.cpp:516
```

**Version**
Commit c0ce5e5

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

**Airframe type**
Tracked rover

**Hardware type**
RPi5 + [PilotPi](https://docs.px4.io/main/en/flight_controller/raspberry_pi_pilotpi.html) + [my patches](https://github.com/ArduPilot/ardupilot/pull/28247/files) (static build)

I'm not too familiar with ardupilot and would appreciate any hint on how to further debug this.
(But of course I hope @andyp1per and/or @tridge can see the cause of the problem at a first glance 😃  ).

heeplr, 2025-03-20:
So far I found that this patch fixes the issue for me:

```diff
diff --git a/libraries/AP_InertialSensor/AP_InertialSensor_Invensensev3.cpp b/libraries/AP_InertialSensor/AP_InertialSensor_Invensensev3.cpp
index daf54ed64e..37fd2b0526 100644
--- a/libraries/AP_InertialSensor/AP_InertialSensor_Invensensev3.cpp
+++ b/libraries/AP_InertialSensor/AP_InertialSensor_Invensensev3.cpp
@@ -185,12 +185,12 @@ AP_InertialSensor_Invensensev3::~AP_InertialSensor_Invensensev3()
 #if HAL_INS_HIGHRES_SAMPLE
     if (highres_sampling) {
         if (fifo_buffer != nullptr) {
-            hal.util->free_type(fifo_buffer, INV3_FIFO_BUFFER_LEN * INV3_HIGHRES_SAMPLE_SIZE + 1, AP_HAL::Util::MEM_DMA_SAFE);
+            hal.util->free_type(fifo_buffer, INV3_FIFO_BUFFER_LEN * INV3_HIGHRES_SAMPLE_SIZE, AP_HAL::Util::MEM_DMA_SAFE);
         }
     } else
 #endif
     if (fifo_buffer != nullptr) {
-        hal.util->free_type(fifo_buffer, INV3_FIFO_BUFFER_LEN * INV3_SAMPLE_SIZE + 1, AP_HAL::Util::MEM_DMA_SAFE);
+        hal.util->free_type(fifo_buffer, INV3_FIFO_BUFFER_LEN * INV3_SAMPLE_SIZE, AP_HAL::Util::MEM_DMA_SAFE);
     }
 }
 
@@ -358,10 +358,10 @@ void AP_InertialSensor_Invensensev3::start()
     // allocate fifo buffer
 #if HAL_INS_HIGHRES_SAMPLE
     if (highres_sampling) {
-        fifo_buffer = hal.util->malloc_type(INV3_FIFO_BUFFER_LEN * INV3_HIGHRES_SAMPLE_SIZE + 1, AP_HAL::Util::MEM_DMA_SAFE);
+        fifo_buffer = hal.util->malloc_type(INV3_FIFO_BUFFER_LEN * INV3_HIGHRES_SAMPLE_SIZE, AP_HAL::Util::MEM_DMA_SAFE);
     } else
 #endif
-    fifo_buffer = hal.util->malloc_type(INV3_FIFO_BUFFER_LEN * INV3_SAMPLE_SIZE + 1, AP_HAL::Util::MEM_DMA_SAFE);
+    fifo_buffer = hal.util->malloc_type(INV3_FIFO_BUFFER_LEN * INV3_SAMPLE_SIZE, AP_HAL::Util::MEM_DMA_SAFE);
 
     if (fifo_buffer == nullptr) {
         AP_HAL::panic("Invensensev3: Unable to allocate FIFO buffer");
@@ -565,7 +565,7 @@ void AP_InertialSensor_Invensensev3::read_fifo()
         tfr_buffer[0] = reg_data | BIT_READ_FLAG;
         // transfer will also be sending data, make sure that data is zero
         memset(tfr_buffer + 1, 0, n * fifo_sample_size);
-        if (!dev->transfer(tfr_buffer, n * fifo_sample_size + 1, tfr_buffer, n * fifo_sample_size + 1)) {
+        if (!dev->transfer(tfr_buffer, n * fifo_sample_size, tfr_buffer, n * fifo_sample_size)) {
             goto check_registers;
         }
         samples = tfr_buffer + 1;
```

tpwrules, 2025-03-31:
Should be fixed by #29164 .

There was some discussion and testing in Discord of Andy's patch in that PR and heeplr confirmed that only Andy's patch fixes it, but we might wish to wait for them to confirm after merging before closing the issue.

heeplr, 2025-04-01:
I can confirm that with this patch, initialization doesn't freeze anymore on RPi5.

tpwrules, 2025-04-05:
Can you check that this is fixed on master?

heeplr, 2025-04-09:
> Can you check that this is fixed on master?

Just checked. Everything working as expected. 👍 
Thanks a lot everyone! Closing this.

closed 2025-04-09 by heeplr