ArduPilot/ardupilot #29594: Setting SIM_PLD_ENABLE Causes PANIC

wyunc, 2025-03-23:
## Bug report

When I set param set [SIM_PLD_ENABLE ](https://ardupilot.org/copter/docs/parameters.html#sim-pld-enable-preland-device-sim-enable-disable)1 in SITL, it causes the copter to disconnect, and the error message is as follows:

PANIC: Should not be called on invalid location: Location cannot be (0, 0, 0)

**Version**
4.7.0

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

wyunc, 2025-03-23:
Additionally, I would like to ask another question. Is the error in [this link](https://discuss.ardupilot.org/t/red-garbled-text-in-console/131995) normal? I'm not entirely sure if it counts as a bug.

peterbarker, 2025-05-01:
> Additionally, I would like to ask another question. Is the error in [this link](https://discuss.ardupilot.org/t/red-garbled-text-in-console/131995) normal? I'm not entirely sure if it counts as a bug.

Yes, that's line noise - we treat malformed mavlink packjets as such and emit them

peterbarker, 2025-05-01:
I've created https://github.com/ArduPilot/ardupilot/pull/29949 to close this issue.

closed 2025-05-02 by peterbarker