ArduPilot/ardupilot #29606: DO NOT make auto yaw when “EKF3 IMU1 MAG0 IN-FLIGHT YAW ALIGNMENT”

lida2003, 2025-03-25:
**Issue details**

EKF3 in flight yaw alignment will adjust the copter heading when in acro.

Detailed disscussion here: https://discuss.ardupilot.org/t/ekf3-imu1-mag0-in-flight-yaw-alignment-makes-auto-yaw-when-takeoff/129629

As it's in Acro mode, we want the system parameters to be aligned, **not** the actual flight heading.  

Our expectations are:  
1. Align parameters such as heading in EKF3.  
2. Maintain the current heading, as it's an Acro flight.

**Version**

4.5.6 copter

**Hardware type**

H743 FC

**Logs**
https://drive.google.com/file/d/1xT8AHv7rP1QzhwMfhur2-BiSFknj6yX5/view?usp=sharing

rmackay9, 2025-03-25:
Thanks for the report, we should be resetting the target heading so that the yaw reset has no effect.  I guess you're saying this is not the case so we need to fix that, thanks!

lida2003, 2025-03-26:
https://github.com/ArduPilot/ardupilot/blob/c09c4621f132a255eb23279e790f339daf58631b/libraries/AP_NavEKF3/AP_NavEKF3_MagFusion.cpp#L118

https://github.com/ArduPilot/ardupilot/blob/c09c4621f132a255eb23279e790f339daf58631b/libraries/AP_NavEKF3/AP_NavEKF3_MagFusion.cpp#L144

https://github.com/ArduPilot/ardupilot/blob/c09c4621f132a255eb23279e790f339daf58631b/libraries/AP_NavEKF3/AP_NavEKF3_PosVelFusion.cpp#L601

It seems that EKF3 performed heading alignment by itself without considering the current flight mode.

rmackay9, 2025-03-26:
Hi @lida2003,

I think the issue is not within the EKF3, it's actually that the AC_AttitudeControl or the mode_acro.cpp code is not handling the yaw reset.  So I think you'll see that if the vehicle is in stabilize mode that the yaw jump does not occur.